libimobiledevice/libimobiledevice #1166: Unable to link libimobiledevice in arch linux.

rohit713, 2021-05-17:
Hello I am a newbie in libimobiledevice. I have installed all libimobiledevice library to my arch linux system. When I am trying to compile my project after linking these library to my qt project. It gives me following error-

**/usr/bin/ld: cannot find -limobiledevice
/usr/bin/ld: cannot find -lplist
/usr/bin/ld: cannot find -lusbmuxd
/usr/bin/ld: cannot find -lirecovery
collect2: error: ld returned 1 exit status**

Can anyone help me to solve this.

mexmer, 2021-05-17:
you need to install development packages for all libraries you missing

eg. libplist-dev, libusbmuxd-dev, libimobiledevice-dev,libirecovery-dev

mind i don't know how exactly they are named on arch, but this is usuall name in other distribution, sometimes also contains version number

rohit713, 2021-05-18:
I could not find any dev packages of libimobiledevice. I installed these library from GitHub. But apart from these I am unable to get it from anywhere. Any more suggestion will be appreciated.

mexmer, 2021-05-18:
if you building from source (github) you need compile a install them in order, also you need to have pkgtool on your system (for configuration)

if i remember correctly, build install order is

libplist
libusbmuxd
libirecovery
libimobiledevice

on each you run
./autogen.sh
make
make install

before compiling next one

nikias, 2021-05-18:
> if you building from source (github) you need compile a install them in order, also you need to have pkgtool on your system (for configuration)
> 
> if i remember correctly, build install order is
> 
> libplist
> libusbmuxd
> libirecovery
> libimobiledevice
> 
> on each you run
> ./autogen.sh
> make
> make install # might need sudo
`sudo ldconfig`

mexmer, 2021-05-18:
ah right, i forgot about that one, i usually build from source under root

rohit713, 2021-05-18:
@nikias , @mexmer I did the same, I installed all the libraries in the order under root user privileges. But again same error come to my project. But I have append(postfix) the version number after the library name. Now its working. I am using below lib name in starting-
**LIBS +=  -ludev -limobiledevice -lplist -lcurl -lzip -lirecovery -lz -lcrypto -lusbmuxd**

and it was giving me the below error-
_/usr/bin/ld: cannot find -limobiledevice
/usr/bin/ld: cannot find -lplist
/usr/bin/ld: cannot find -lusbmuxd
/usr/bin/ld: cannot find -lirecovery
collect2: error: ld returned 1 exit status_
now I am using the below library name-
**LIBS +=  -ludev -limobiledevice-1.0 -lplist-2.0 -lcurl -lzip -lirecovery-1.0 -lz -lcrypto -lusbmuxd-2.0**
It works fine.
But I am little-bit confused for future update of libimobiledevice. I think I will have to update these library version with my qt project binary ? @nikias

nikias, 2021-05-18:
Ah of course. Yes you have to append the API version to the lib name, so `-lplist-2.0`, `-limobiledevice-1.0` etc.

rohit713, 2021-05-18:
@nikias Ok I'll do. And thank you so much for the reply.

closed 2021-05-18 by rohit713