libimobiledevice/libimobiledevice #1172: Get replaced hardware components like 3uTools ?

Adderally, 2021-05-23:
Any way of telling if any hardware has been substituted on the device, Eg: The Display ?

truonggiang0710, 2021-06-24:
3uTools can do this, but it not support check screen serial number (Display).

Schlaubischlump, 2021-11-14:
If you take a look at the [libMobileGestalt](https://pastebin.com/HY122aPR) keys there are a couple of keys containing the word serial number. E.g 'RawPanelSerialNumber', 'PanelSerialNumber' or 'AmbientLightSensorSerialNumber'. Some of them might be related to the serial number of a display component. Given the serial number you can [check the manufacturer](https://www.iparts-4u.co.uk/blog/how-to-identify-and-copy-iphone-screen-data/) and might be able to determine based on this information if the screen was replaced.

msk521, 2021-12-12:
> If you take a look at the [libMobileGestalt](https://pastebin.com/HY122aPR) keys there are a couple of keys containing the word serial number. E.g 'RawPanelSerialNumber', 'PanelSerialNumber' or [...]

when i use RawPanelSerialNumber、PanelSerialNumber