libimobiledevice/libimobiledevice #1173: WatchOS support

mexmer, 2021-05-24:
seems there is issue with lockdown handshake on WatchOs, i have applewatch connected trough usb adapter and when it comes to handshake, it tries to do validate_pair and fails with unknown error (tested on watchos 6.3 and 7.1), i will do further research on it, and try to make fix for it, if i skip validate_pair, it works properly it seems.

it my own code, that communicates with watch trough ioctl driver (on windows) i skip validate_pair completely for iwatch, but since there is no detection on libimobile if device is iwatch and only version is compared for validate pair, it is not skipped.

you can query "DeviceClass" which from my testing contains following values "Watch","iPad","iPhone","AppleTV" ... maybe some other value for mac computers, but i had not tested those.

mexmer, 2021-05-24:
adding proposed patch.
mind i don't have older device than watch3 with os 6.3/7.1, so can't confirm functionality on older WatchOS versions

```diff
Index: idevice.c
===================================================================
--- idevice.c	(revision 27887)
+++ idevice.c	(working copy)
@@ -355,6 +355,7 @@
 	device->udid = strdup(muxdev->udid);
 	device->mux_id = muxdev->handle;
 	device->version = 0;
+	device->device_class = NULL;
 	switch (muxdev->conn_type) {
 	case CONNECTION_TYPE_USB:
 		device->conn_type = CONNECTION_USBMUXD;
@@ -412,6 +413,7 @@
 	ret = IDEVICE_E_SUCCESS;
 
 	free(device->udid);
+	free(device->device_class);
 
 	if (device->conn_data) {
 		free(device->conn_data);
Index: idevice.h
===================================================================
--- idevice.h	(revision 27887)
+++ idevice.h	(working copy)
@@ -66,6 +66,7 @@
 	enum idevice_connection_type conn_type;
 	void *conn_data;
 	int version;
+	char* device_class;
 };
 
 #endif
Index: lockdown.c
===================================================================
--- lockdown.c	(revision 27887)
+++ lockdown.c	(working copy)
@@ -699,6 +699,18 @@
 		plist_free(p_version);
 	}
 
+	if (device->device_class == NULL) {
+		plist_t p_dev_class = NULL;
+		if (lockdownd_get_value(client_loc, NULL, "DeviceClass", &p_dev_class) == LOCKDOWN_E_SUCCESS) {
+			int vers[3] = { 0, 0, 0 };
+			char* s_dev_class = NULL;
+			plist_get_string_val(p_dev_class, &s_dev_class);
+			if (s_dev_class != NULL)
+				device->device_class = s_dev_class;
+		}
+		plist_free(p_dev_class);
+	}
+
 	userpref_read_pair_record(client_loc->udid, &pair_record);
 	if (pair_record) {
 		pair_record_get_host_id(pair_record, &host_id);
@@ -715,7 +727,7 @@
 	plist_free(pair_record);
 	pair_record = NULL;
 
-	if (device->version < DEVICE_VERSION(7,0,0)) {
+	if (device->version < DEVICE_VERSION(7,0,0) && (device->device_class == NULL || strcmp(device->device_class, "Watch"))) {
 		/* for older devices, we need to validate pairing to receive trusted host status */
 		ret = lockdownd_validate_pair(client_loc, NULL);
```

a-gunter, 2021-07-23:
Confirmed this works on different series Watches and different WatchOS versions, thanks!

nikias, 2021-07-29:
Added with commit ec8084b075c56380806c5c79e3e5db039c8adc90 in a slightly different variant.

closed 2021-07-29 by nikias

mexmer, 2021-07-30:
yeah, looks better, thanks.

truonggiang0710, 2021-11-25:
Hi @mexmer/@a-gunter , can the libimobiledevice connect Apple watch S6 through a USB adaptor on Ubuntu ? When I connect Apple watch S6, the idProduct = 0x1881, usbmuxd does not recognize the watch.

mexmer, 2021-11-25:
> Hi @mexmer/@a-gunter , can the libimobiledevice connect Apple watch S6 through a USB adaptor on Ubuntu ? When I connect Apple watch S6, the idProduct = 0x1881, usbmuxd does not recognize the watch.

not sure what adapter you using, but if you using iBus adapter (which i have tested)
https://www.mfcbox.com/shop/ibus-s4-5-tool-for-apple-watch-s4-s5-40mm-44mm/
then no , it doesn't work, this adapter doesn't propagate mux service, i have tested it with windows and linux, and does propagate some interface, but it is not recognized and i was not able to communicate with it either trough libusb directly or trough 

i tested iBus adapters (S1/S2/S4) with all apple watch devices up to watch 6, and all worked except 6
watch series 0, 1 - iBus S1 - will connect, mux interface exposed, works
watch series 2, 3 - iBus S2 - will connect, mux interface exposed, works
watch series 4, 5, SE - iBus S4/5 - will connect, mux interface exposed, works
watch series 6 - iBus S4/5 - will connect, no mux interface, doesn't communicate

as you have noticed series 6 propagate with different USB product id, while other watches all use same USB product id. i tried few times to contact author of iBus adapter, but never got any response.

truonggiang0710, 2021-11-25:
Thanks @mexmer , I requested the MFC team to supply the driver for S6 on Ubuntu and windows, but the answer is still "waiting".
I have an S6 driver on Mac. The Apple watch S6 connects successfully to the Mac. It can read information and restore with ipsw file. 
I capture the data. The header of sending package adds 28 bytes before usbmuxd header. I don't know what the data.
Below a sample of header.
**4F 10 A0 10 03 0D 00 00 54 01 00 00 AC D5 25 00
00 00 00 00 48 01 00 00 44 01 00 11** 00 00 00 06
00 00 01 44 FE ED FA CE 01 BF 00 FC 33 00 F2 7E
00 00 00 01 00 00 00 01 5E 10 02 00 2E 30 22 20
00 00 01 1C 3C 3F 78 6D 6C 20 76 65 72 73 69 6F
6E 3D 22 31 2E 30 22 20 65 6E 63 6F 64 69 6E 67
3D 22 55 54 46 2D 38 22 3F 3E 0A 3C 21 44 4F 43
54 59 50 45 20 70 6C 69 73 74 20 50 55 42 4C 49
43 20 22 2D 2F 2F 41 70 70 6C 65 2F 2F 44 54 44
20 50 4C 49 53 54 20 31 2E 30 2F 2F 45 4E 22 20
22 68 74 74 70 3A 2F 2F 77 77 77 2E 61 70 70 6C
65 2E 63 6F 6D 2F 44 54 44 73 2F 50 72 6F 70 65
72 74 79 4C 69 73 74 2D 31 2E 30 2E 64 74 64 22
3E 0A 3C 70 6C 69 73 74 20 76 65 72 73 69 6F 6E
3D 22 31 2E 30 22 3E 0A 3C 64 69 63 74 3E 0A 09
3C 6B 65 79 3E 4C 61 62 65 6C 3C 2F 6B 65 79 3E
0A 09 3C 73 74 72 69 6E 67 3E 69 64 65 76 69 63
65 69 6E 66 6F 3C 2F 73 74 72 69 6E 67 3E 0A 09
3C 6B 65 79 3E 52 65 71 75 65 73 74 3C 2F 6B 65
79 3E 0A 09 3C 73 74 72 69 6E 67 3E 51 75 65 72
79 54 79 70 65 3C 2F 73 74 72 69 6E 67 3E 0A 3C
2F 64 69 63 74 3E 0A 3C 2F 70 6C 69 73 74 3E 0A

Regards.

mexmer, 2021-11-25:
yes i know, it can be used to write IPSW (firmware) to device, but it can't be used for anything else. with modified driver, exposed interface reports as apple debug interface .... but there is no mux endpoint or at least i was not able to find proper configuration, to enable it.

truonggiang0710, 2021-11-26:
I hope we can find out the proper configuration.
Happy Thanksgiving to you and libimobiledevice team that is always doing great things. You are wonderful, and I hope you have a great holiday with the people in your life who matter most.
Thanks.

mexmer, 2021-11-29:
@truonggiang0710 i'm only libimobile user, working on commercial solution based around libimobile, so you should direct your thanks more likely to @FunkyM , @nikias and rest of libimobile team 👋

jsonlee1688, 2024-07-18:
@mexmer How do you access iwatch with libimobiledevice?  By a usb cable or via wifi network? I used my charger cable but no working, could you please help? thank you!

mexmer, 2024-07-18:
as mentioned in [here](#issuecomment-978914997) , you need special reader (adapter)
as for newer watches that do not have service connector(S7,S8,S9,Ultra, SE2), you need special cradle for wireless acces, it doesn't work under windows or linux tho', on mac it does work trough service in OS X ... S6 watches also work on mac with iBus adapter (but not on windows or linux)

gsmdenis, 2024-07-30:
for new watch S7,S8,S9,Ultra, SE2 need https://www.mfcbox.com/shop/ibus-x-tool-for-apple-watch-s7-s8-s9-ultra-and-ultra2

Malinacek, 2025-09-10:
> for new watch S7,S8,S9,Ultra, SE2 need https://www.mfcbox.com/shop/ibus-x-tool-for-apple-watch-s7-s8-s9-ultra-and-ultra2

SE 2 still has a physical diagnostic port