libimobiledevice/libimobiledevice #1186: I need android example

tangwanchao, 2021-07-12:
I have successfully run the library on Android, but after connecting the IOS device, usbmuxd_get_device_list tells me that there is no device connection, I don't know what to do, and need your help

hcropps, 2022-09-13:
@tangwanchao I have the same problem. Did you solve the problem?

xanadu-lee, 2023-05-16:
The reason is that we need to run usbmuxd

PeiNiWan1, 2024-03-06:
在android 中无法扫描到usb设备 usbmuxd_get_device_list结果为：-2

AiXanadu, 2024-03-07:
@PeiNiWan1 因为libusb要进行一些修改。

PeiNiWan1, 2024-03-07:
> @PeiNiWan1 因为libusb要进行一些修改。
我需要将该库移植到Android平台，但是对于该方向不了解，能否联系你，提供有偿协助，哥

AiXanadu, 2024-03-07:
@PeiNiWan1 你是哪家公司喔，这涉及到公司机密，可能我无法有效提供帮助。

PeiNiWan1, 2024-03-07:
个人使用，在校大学生，可以提供学生证明，毕业设计

PeiNiWan1, 2024-03-07:
> @PeiNiWan1 你是哪家公司喔，这涉及到公司机密，可能我无法有效提供帮助。

个人使用，在校大学生，可以提供学生证明，毕业设计,哥，了解了解 给个方向也行

AiXanadu, 2024-03-07:
> > @PeiNiWan1 你是哪家公司喔，这涉及到公司机密，可能我无法有效提供帮助。
> 
> 个人使用，在校大学生，可以提供学生证明，毕业设计,哥，了解了解 给个方向也行，实在没办法才来寻求协助的

我只能说这个需要了解NDK，Java，Android，以及C++。才可以弄。个人建议别研究了。

closed 2024-03-29 by tangwanchao