libimobiledevice/libimobiledevice #1189: idevicebackup2 broken on Big Sur 11.5?

marcinx, 2021-07-22:
Started running into this when trying to run a backup:

```
Started "com.apple.mobilebackup2" service on port 50343.
Could not perform backup protocol version exchange, error code -1
```

It was working fine before updating to Big Sur 11.5. Tried against two devices with iOS 14.7 and one device with iOS 14.2. On all three devices `idevicebackup2` terminates with above error. `ideviceinfo` works fine. Haven't tried other commands.

Any hints?

marcinx, 2021-07-22:
Installed libimobiledevice on Catalina 10.15.4 and curiously enough am getting a similar error:
```
Started "com.apple.mobilebackup2" service on port 49417.
Could not perform backup protocol version exchange, error code -1
```

mexmer, 2021-07-22:
you have most likely installed old version of libimobiledevice trough brew, you need to compile libimobile from source

nikias, 2021-07-24:
Yes, old version. Please use my script to build everything directly from git. In case you have it installed through homebrew or macports, uninstall them first.
```bash
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -Ls https://git.io/J3kfy
bash ./limd-build-macos.sh
```

marcinx, 2021-07-24:
Thanks. Yes. It was a brew install. Will try. It was odd that it worked at first and then stopped.

closed 2021-07-24 by marcinx