libimobiledevice/libimobiledevice #1190: Package requirements (libplist-2.0 >= 2.2.0) were not met:

fsinnes, 2021-07-23:
I may be a fucking idiot. I'm new to this whole scenario but I've learned alot in the past week.

I am sorry if this is too much code but this is starting with the libplist clone.

previous steps with no error or issue

(using msys2 32bit)

mkdir build
cd build/
pacman -S mingw-w64-i686-toolchain
pacman -S --needed base-devel msys2-devel
pacman -S openssl-devel
pacman -S libreadline-devel
pacman -S mingw-w64-i686-libusb
pacman -S mingw-w64-i686-libzip
pacman -S mingw-w64-i686-libxml2
pacman -S mingw-w64-i686-libssh2
pacman -S mingw-w64-i686-curl
pacman -S git

then....

$ git clone https://github.com/libimobiledevice/libplist.git
Cloning into 'libplist'...
remote: Enumerating objects: 4513, done.
remote: Counting objects: 100% (112/112), done.
remote: Compressing objects: 100% (75/75), done.
remote: Total 4513 (delta 55), reused 72 (delta 31), pack-reused 4401
Receiving objects: 100% (4513/4513), 1.36 MiB | 9.72 MiB/s, done.
Resolving deltas: 100% (2782/2782), done.
Updating files: 100% (205/205), done.

MET DEV@Forrest-PC MINGW32 ~/build
$ cd libplist

MET DEV@Forrest-PC MINGW32 ~/build/libplist
$ ./autogen.sh
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:23: installing './compile'
configure.ac:36: installing './config.guess'
configure.ac:36: installing './config.sub'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
cython/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
configure.ac:36: warning: The macro `AC_PROG_LIBTOOL' is obsolete.
configure.ac:36: You should run autoupdate.
m4/libtool.m4:99: AC_PROG_LIBTOOL is expanded from...
configure.ac:36: the top level
configure.ac:39: warning: The macro `AC_HEADER_STDC' is obsolete.
configure.ac:39: You should run autoupdate.
../autoconf-2.71/lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
configure.ac:39: the top level
configure.ac:70: warning: $as_echo is obsolete; use AS_ECHO(["message"]) instead
../autoconf-2.71/lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
../autoconf-2.71/lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
../autoconf-2.71/lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
../autoconf-2.71/lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/ax_pthread.m4:88: AX_PTHREAD is expanded from...
configure.ac:70: the top level
configure.ac:80: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:80: You should run autoupdate.
[... 484 lines cut ...]
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking whether the g++ linker (C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe) supports shared libraries... yes
checking for g++ option to produce PIC... -DDLL_EXPORT -DPIC
checking if g++ PIC flag -DDLL_EXPORT -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe) supports shared libraries... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /mingw32/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libplist... no
configure: error: Package requirements (libplist-2.0 >= 2.2.0) were not met:

Package 'libplist-2.0', required by 'virtual:world', not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libplist_CFLAGS
and libplist_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

TheCabl3Gui, 2021-07-28:
some might follow the logical steps of the install wiki portion by sudo apt-get libimobiledevice and pulling the needed clones, but proceeding TO NOT cd to the clones and make them before making libimobiledevice.

The takeaway from this is:
 
LIBPLIST is essential for LIBUSBMUXD and they are both needed for LIBIMOBILEDEVICE.

to avoid this problem simply follow the steps: 

1. where you should be: home directory
```````bash`````````````````````````````````````````````````
sudo apt-get install \ build-essential \
	checkinstall \
	git \
	autoconf \
	automake \
	libtool-bin
```````bash``````````````````````````````````````````````````
2. where should you start? home directory
what are you doing? Getting a clone and changing the directory to it.
``````````bash````````````````````````````````````````````````
>cd
>git clone https://github.com/libimobiledevice/libplist.git
>cd libplist

3. in the same terminal and directory: [IE from user@device~/libplist:]
```````````````bash```````````````````````````````````````````````
>./autogen.sh
>make
>sudo make install
`````````````````````````````````````````````````````````````````````
4. change to home directory and git clone of libusbmuxd:
`````bash`````````````````````````````````````````````````````````
>cd
>git clone https://github.com/libimobiledevice/libusbmuxd.git
>cd libusbmuxd
````````````````````````````````````````````````````````````````````
5. Now we are going to build, make and install libusbmuxd with these bash commands from libusbmuxd directory:
`````bash````````````````````````````````````````
>./autogen.sh
>make
>sudo make install
``````````````````````````````````````````````````
6. Now that you have libplist && libusbmuxd you can make libimobiledevice:
IRL redo the install from the wiki

nikias, 2021-07-28:
I suggest to also run `sudo ldconfig` after each `sudo make install` step.

TheCabl3Gui, 2021-07-28:
how embarrassing for me lol. well good luck. i need to read the whole thing before commenting. Moreover, it would be good to know that libimobiledevice is dependent on libplist and libusbmuxd in that order. have you considered closing everything and rebooting. then touch all the dependents.

nikias, 2021-07-28:
@TheCabl3Gui no worries. The title of the issue doesn't say it's about Windows. I also only realized after I posted my reply :D 
I am not sure what you mean with "closing everything"? Anyhow I think we can update the instructions to make this more clear.

fsinnes, 2021-07-29:
So the issue is.. if I remember correctly...
the current version is 2.2.1 (which is what installs) but libusbmuxd wont build with "./autogen.sh" because I have a newer version than what it searches for
again... I'm not a "pro"grammer... so I'm sorry if I sound stupid
I tried cloning an older commit of 2.2.0 recently and that didnt work.
I have tried "closing everything" I guess by completely removing everything related to MSYS and started over with same results
I THINK its just because I need to wait for an updated version of libusbmuxd

pfidelus-dev, 2024-07-09:
same here, any updates on this?