libimobiledevice/libimobiledevice #1191: ideviceinfo displays incorrect device information for iPhone 11

sr-murthy, 2021-07-23:
ideviceinfo (1.3.1) is displaying incorrect device information about an iPhone 11 (running iOS 14.7). To be precise, it is reporting that the device, while connected, is an iPhone 12 (False), is not password protected (False), does not use a 24 hour block (False). It is also falsely reporting the MAC address, and falsely reporting the phone number - it displays the phone number associated with the temporary SIM bundled with the phone, not the SIM from my previous iPhone, which is the one that is now associated with the phone.

nikias, 2021-07-23:
Before assuming values are false, you need to understand what the values actually mean, and where they come from.
The device provides the data, and `ideviceinfo` just prints what the device returns.

Some explanations and tips:
- `ideviceinfo -k ProductType` will show `iPhone12,1` for the iPhone 11, because that's what Apple calls them internally. The ideviceinfo output does not show the marketing name.
- `ideviceinfo -k PasswordProtected` shows the current lock status, not if the device has a passcode set or not. Use `idevicediagnostics mobilegestalt PasswordConfigured` to show if a passcode is set or not.
- `Uses24HourClock` seems indeed not reflecting the system settings. But again, the device reports this and it looks like Apple abandoned this setting/lockdown key. I don't see a replacement for this right away.
- Regarding the MAC address, make sure to pick the right one, as there are 3: `BluetoothAddress`, `EthernetAddress`, and `WiFiAddress`: `ideviceinfo |grep -i address`
- Phone number is user-configurable in settings: Settings -> Phone -> My Number` and this setting will correctly be shown in ideviceinfo output. If this number is set and you change SIM cards, this setting will NOT be updated automatically from what I have experienced.

closed 2021-07-23 by nikias

a-gunter, 2021-07-23:
Just a note...

I checked with both iPhone 11 and iPhone 12 Pro Max (both on iOS 14.7) and ideviceinfo is correctly reporting "Uses24HourClock" to be synced with the value in Settings -> General -> Date & Time -> 24-Hour Time

sr-murthy, 2021-07-24:
Thanks for the clarifications.