libimobiledevice/libimobiledevice #1192: Possible problems with libimobiledevice backup or backup decryption

sr-murthy, 2021-07-23:
I'm using the iOS mvt `check-backup` command to analyse the decrypted backup of an iPhone 11 (running 14.7), but getting the following errors in the log relating to the `Manifest.db` produced by the backup decryption. The iPhone backup was done with `idevicebackup2 backup --full /path/to/target/`. If the database file is damaged that might indicate problems with the backup decryption or the backup itself. Is this a known problem, or worth investigating?
```
ERROR    [mvt.ios.modules.fs.net_datausage] Error in running extraction from module Datausage: database disk image is    
                  malformed                                                                                                       
                  Traceback (most recent call last):                                                                              
                    File "/usr/local/lib/python3.8/site-packages/mvt/common/module.py", line 135, in run_module                   
                      module.run()                                                                                                
                    File "/usr/local/lib/python3.8/site-packages/mvt/ios/modules/fs/net_datausage.py", line 29, in run            
                      self._extract_net_data()                                                                                    
                    File "/usr/local/lib/python3.8/site-packages/mvt/ios/modules/fs/net_base.py", line 20, in _extract_net_data   
                      cur.execute("""SELECT                                                                                       
                  sqlite3.DatabaseError: database disk image is malformed
...
ERROR    [mvt.ios.modules.fs.sms] Error in running extraction from module SMS: database disk image is malformed          
                  Traceback (most recent call last):                                                                              
                    File "/usr/local/lib/python3.8/site-packages/mvt/common/module.py", line 135, in run_module                   
                      module.run()                                                                                                
                    File "/usr/local/lib/python3.8/site-packages/mvt/ios/modules/fs/sms.py", line 57, in run                      
                      cur.execute("""                                                                                             
                  sqlite3.DatabaseError: database disk image is malformed
...
ERROR    [mvt.ios.modules.fs.sms_attachments] Error in running extraction from module SMSAttachments: database disk image
                  is malformed                                                                                                    
                  Traceback (most recent call last):                                                                              
                    File "/usr/local/lib/python3.8/site-packages/mvt/common/module.py", line 135, in run_module                   
                      module.run()                                                                                                
                    File "/usr/local/lib/python3.8/site-packages/mvt/ios/modules/fs/sms_attachments.py", line 45, in run          
                      cur.execute("""                                                                                             
                  sqlite3.DatabaseError: database disk image is malformed
...
ERROR    [mvt.ios.modules.fs.calls] Error in running extraction from module Calls: database disk image is malformed      
                  Traceback (most recent call last):                                                                              
                    File "/usr/local/lib/python3.8/site-packages/mvt/common/module.py", line 135, in run_module                   
                      module.run()                                                                                                
                    File "/usr/local/lib/python3.8/site-packages/mvt/ios/modules/fs/calls.py", line 41, in run                    
                      cur.execute("""                                                                                             
                  sqlite3.DatabaseError: database disk image is malformed 
```

nikias, 2021-07-23:
Check the `Manifest.plist` in `backupfolder/DEVICE_UDID/Manifest.plist` has a key `IsEncrypted`. Make sure it is false. If not, the backup is encrypted. Use `plistutil` to convert the file to XML (or use `plutil -p` on it if you're on macOS).

sr-murthy, 2021-07-24:
The original backup was encrypted. I turned off the encryption option, and the errors disappeared. Thanks.

Note that the MVT docs on `libimobiledevice` (https://mvt.readthedocs.io/en/latest/ios/backup/libimobiledevice.html) say that

> encrypted backup contain more data than unencrypted backups

I'm not clear on whether this is simply an observation, or a recommendation.

closed 2021-07-24 by sr-murthy

nikias, 2021-07-24:
@sr-murthy I guess reading their docs can be helpful, MVT supports backup decryption:
https://mvt.readthedocs.io/en/latest/ios/backup/check.html