libimobiledevice/libimobiledevice #1193: idevicebackup2 isn't sending?

JJedrasik, 2021-07-25:
Hi guys,
I'm trying to create a backup file for my iphone (ios 14.6). Device is found and recognized, no issues, I can turn encryption on and off, no issues, but when I try to run a backup, i.e.
`idevicebackup2 -d backup --full iphonebackup/`
The output resolves to
`Backup directory is "iphonebackup/"
Started "com.apple.mobilebackup2" service on port 64771
Negotiated Protocol Version 2.1
Reading Info.plist
Starting backup.....
Enforcing full backup from device.
Backup will be unecrypted.
Requested backup from device.....
Full backup mode.
`

And then it just kinda sits there, I let it sit even for over an hour, and nothing. 

One time it did spit out an error at me (105), which is caused by insufficient free disk space...my phone has 143GB out of 256 used, maybe thats too much?

nikias, 2021-07-25:
It can take some time to start, but it shouldn't be over an hour. Launch `idevicesyslog -p BackupAgent2` in another terminal to see what is going on.

JJedrasik, 2021-07-25:
The syslog looks exactly like how i'd expect, looking for things to backup, all <Notice> tags. Oddly enough, it did eventually stop and said 

`Insufficient free disk space on drive to backup`

This VM only has a 40GB drive, while my iphone has about 143GB of stuff, maybe this is the problem?

nikias, 2021-07-25:
Yes. Definitely a problem. Where would it store all the data?

JJedrasik, 2021-07-26:
Yeah that fixed it, i'm an idiot - closing this now

closed 2021-07-26 by JJedrasik