libimobiledevice/libimobiledevice #1197: Unable to connect to iPhone 12 w/ iOS 14.7.1 on Ubuntu 20.04

bviktor, 2021-08-04:
So I've built packages for:

- libplist3
- libusbmuxd6
- libimobiledevice6
- usbmuxd

From the latest Git commits, ran `ldconfig`, `systemctl daemon-reload`, `udevadm control  --reload-rules`, `udevadm trigger --action=change`, then tried to connect my phone. The phone asks me if I trust the computer, I say I do, enter my passcode, and I receive a GNOME notification too that I've connected my iPhone.

This is the usbmuxd output:

```
- Logs begin at Thu 2021-05-20 17:35:44 CEST. --
Aug 03 11:04:39 foo001linux.ad.foobar.com usbmuxd[17818]: [11:04:39.770][3] Removed device 1 on location 0x10005
Aug 03 11:04:39 foo001linux.ad.foobar.com usbmuxd[17818]: [11:04:39.777][3] usbmuxd shutting down
Aug 03 11:04:39 foo001linux.ad.foobar.com usbmuxd[17818]: [11:04:39.878][3] Shutdown complete
Aug 03 11:04:39 foo001linux.ad.foobar.com systemd[1]: usbmuxd.service: Succeeded.
Aug 04 11:30:20 foo001linux.ad.foobar.com systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Aug 04 11:30:20 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:20.266][3] usbmuxd v1.1.2 starting up
Aug 04 11:30:20 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:20.268][3] Successfully dropped privileges to 'usbmux'
Aug 04 11:30:20 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:20.268][3] Using libusb 1.0.23
Aug 04 11:30:20 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:20.270][3] Initialization complete
Aug 04 11:30:20 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:20.270][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Aug 04 11:30:48 foo001linux.ad.foobar.com usbmuxd[138847]: libusb: error [op_get_configuration] device unconfigured
Aug 04 11:30:48 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:48.352][3] Connecting to new device on location 0x10006 as ID 1
Aug 04 11:30:48 foo001linux.ad.foobar.com usbmuxd[138847]: [11:30:48.352][3] Connected to v2.0 device 1 on location 0x10006 with serial number xxx
Aug 04 11:30:48 foo001linux.ad.foobar.com usbmuxd[138847]: libusb: error [udev_hotplug_event] ignoring udev action bind
```

Now if I try to run `idevicename`, for the first minute or two it says:

```
ERROR: No device found.
```

Then:

```
ERROR: Could not connect to lockdownd, error code -8
```

And that's where things end. Any ideas?

demhademha, 2021-08-04:
Could you use the `-d` flag:
`idevicename -d`
And paste the output

bviktor, 2021-08-04:
Unfortunately it gives the same result. Should I rebuild the packages with `debug`?

nikias, 2021-08-04:
First thing to try is not something like idevicename, but `idevice_id` since that gives us an idea if the connection with usbmuxd works and if it sees the device.

ruffson, 2021-08-28:
Sorry to tag on, but since nothing happened here for a while and I have the same problem with the same OS version and a iPhone 12 mini on Fedora: `idevice_id` I get: `ERROR: Unable to retrieve device list!`.
I ran all the mentioned commands as OP but nothing changed. 

All commands I tried result in the same `ERROR: No device found.` or similar, same as `ifuse`.
Now after I tried reconnecting my phone a few times, when I connect it to my computer on any port, it does not show the "Trust this device" dialog and also does not charge on any port (weirdly sometimes it still makes that "plunk" charging soung). It seems to have (temporarily?) black-listed my notebook? (I have another one I could try if that makes sense)

Any advice how to proceed would be very helpful.

Some journal entries from a little earlier where I believe it still somehow recognized the phone:
```
Aug 27 17:04:03 comp systemd[1]: /usr/lib/systemd/system/usbmuxd.service:7: PIDFile= references a path below legacy directory /var/run/, updating /var/run/usbmuxd.pid → /run/usbmuxd.pid; please update the unit file accordingly.
Aug 27 17:04:03 comp audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=usbmuxd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.037][3] usbmuxd v1.1.1 starting up
Aug 27 17:04:03 comp audit[649383]: AVC avc:  denied  { getattr } for  pid=649383 comm="usbmuxd" name="/" dev="dm-0" ino=256 scontext=system_u:system_r:usbmuxd_t:s0 tcontext=system_u:object_r:fs_t:s0 tclass=filesystem permissive=0
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.039][3] Successfully dropped privileges to 'usbmuxd'
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.039][3] Using libusb 1.0.24
Aug 27 17:04:03 comp usbmuxd[649383]: libusb: warning [op_get_configuration] device unconfigured
Aug 27 17:04:03 comp usbmuxd[649383]: libusb: error [op_get_active_config_descriptor] device unconfigured
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.042][3] Could not get old configuration descriptor for device 1-22: LIBUSB_ERROR_NOT_FOUND
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.112][3] Initialization complete
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.112][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.113][3] Connecting to new device on location 0x10016 as ID 1
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.114][3] Connected to v2.0 device 1 on location 0x10016 with serial number 00008101-0009190911E0001E
Aug 27 17:04:03 comp usbmuxd[649383]: [17:04:03.234][3] Removed device 1 on location 0x10016
Aug 27 17:14:57 comp systemd[1]: /usr/lib/systemd/system/usbmuxd.service:7: PIDFile= references a path below legacy directory /var/run/, updating /var/run/usbmuxd.pid → /run/usbmuxd.pid; please update the unit file accordingly.
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
Aug 27 17:15:16 comp usbmuxd[649383]: libusb: error [udev_hotplug_event] ignoring udev action change
```

I have libusbmuxd, libusbmux-devel and usbmuxd installed.

nikias, 2021-08-29:
`ERROR: Unable to retrieve device list!` means that usbmuxd is either not running or `idevice_id` cannot connect to the usbmuxd socket file which defaults to `/var/run/usbmuxd`. Make sure that file exists after starting usbmuxd. Also in your log output the device is removed right after it was detected, which is a bit weird too. What if you run `usbmuxd` manually? Like `sudo usbmuxd -f -v -v` and then plug in a device?

ruffson, 2021-08-30:
> `sudo usbmuxd -f -v -v`

When I run that, I get:

```
[21:31:57.371][3] usbmuxd v1.1.1 starting up
[21:31:57.371][1] Another instance is already running (pid 649383). exiting.
```
So it seems to be running. But I am still not able to get the id.

nikias, 2021-08-30:
Unplug all devices, then run the command, then plug in a device.

ruffson, 2021-08-30:
> Unplug all devices, then run the command, then plug in a device.

That did not work either but I just restarted the systemd service which resolved all problems:

```
sudo systemctl restart usbmuxd
```

Thanks everyone for the help!