libimobiledevice/libimobiledevice #1199: GetValue returning only simple output

AudiCoder, 2021-08-10:
When trying to call GetValue Api from lockdown. I am getting just the simple output. Is there a way to not use simple when calling the API?

lockdown_test.go:49: map[FamilyID:-1 MinITunesVersion:12.10.10 MinMacOSVersion:10.5.8]

nikias, 2021-08-10:
Not sure what you refer to with "simple output".
Some lockdown values can only be retrieved if the device is paired and you have to start a lockdown session first. Check the code in ideviceinfo.

AudiCoder, 2021-08-10:
@nikias I meant the output when we use "--simple"

Eg:
I want to get some particular key in domain: com.apple.mobile.iTunes which is not present in simple output

Simple Command: ➜  bin ideviceinfo -u 00008020-000221942EEA002E -q com.apple.mobile.iTunes --s     

Output
FamilyID: 18446744073709551615
MinITunesVersion: 12.9.0
MinMacOSVersion: 10.5.8
iTunesSetupComplete: true

I am looking for key which is present after pairing device hence key is not present under simple.
Command: ideviceinfo -u 00008020-000221942EEA002E -q com.apple.mobile.iTunes

When I use lockdown get_value, I see command being executed is with simple. Is there a way that I can do pairing when executing GetValue to get full output?

nikias, 2021-08-10:
You can try to pair it with `idevicepair pair`, but other than that if you don't pass `-s` it will start a lockdown session and request the value.

AudiCoder, 2021-08-12:
I was able to get past by pairing the device first. However, I wanted to go through the second route by not passing "-s" in the API but I do not see an option to toggle "simple" value while calling "get_value" API in lockdown.

nikias, 2021-08-13:
The API does not have simple vs. not simple. Some values are just not available if you don't start a lockdown session. So if you want to query like ideviceinfo does with -s you just don't start a session.

mexmer, 2021-08-13:
in old versions of ios it was possible to get lot of info without session, but since ios 5 apple had gradually gated more and more values behind session, so you could not get info without pairing device first.

on ios 3 you could get like everything without pairing, on current ios, there is not much to be retrieved, you can get model, ios version, udid and few other values that OS X uses to properly detect device.

now you can't even restart phone from computer without pairing it first (changed since ios 14.5)