libimobiledevice/libimobiledevice #1200: Is restoring an encrypted backup working at all?

v0tti, 2021-08-10:
I tried to back up a freshly set up iPad.

- `idevicebackup2 encryption on -i` is successful and will ask for the device password on my iPad, but not on the command line.
- `idevicebackup2 backup --full ./iosbackup` is successfull.
- Reset iPad -> Restore from iTunes.
- `idevicebackup2 -i restore ./iosbackup` asks for a password on the command line. The device password does not work. I do not have any other password.

Running `idevicebackup2 encryption on --password 12345` is successful but does not seem to have any effect on the actual encryption password.

So is it at all possible to do an encrypted backup of a freshly installed iOS device?

nikias, 2021-08-10:
You are using the options incorrectly. With the first command you set the password to `-i`. If you want interactive mode you need to pass it before the command `encryption`. That is also the case with any other option:
`idevicebackup -i encryption on`.
The last command you tried set your password to `--password`.

v0tti, 2021-08-10:
It seems I am an idiot. Thanks for the reply.

closed 2021-08-10 by v0tti