libimobiledevice/libimobiledevice #1201: Any chance for a new release?

jmpolom, 2021-08-13:
Last release was over a year ago and since then there has been an update to ensure backups work with iOS 14. This update is in the master branch in git, but hasn't made it into a release. Any thoughts on when the next release may occur?

cloudlena, 2022-07-17:
@nikias¸ could you please share a release roadmap if there is one? People have been asking for a new release that supports iOS 14 and 15 for more than a year, and it would be wonderful to see it released soon :)

SonGokuSSJ, 2023-01-25:
Yes, would be nice if a new release version could be created for all components. The last release on the release page is from Summer 2020, so about 2.5 years old. Many fixes have been implemented since then (i.e. needed ones to be able to connect to iOS 16).