libimobiledevice/libimobiledevice #1204: idevicebackup2 - What exactly does it create a copy of?

jokerekv2, 2021-08-22:
The backup created with itunes weighs about 20Gb and with idevicebackup2 only 4Gb. I don't really understand how this works. What exactly does it create a copy of? Anyone anything?

mexmer, 2021-08-23:
did you do full backup with idevicebackup?

itunes explictily call full backup, if there is no previous backup present (or if there is big time difference from old backup)

idevicebackup calls service with "default" values ... what is default knows only apple, but if you want to make sure to do "full" backup you need to call with -- full parameter
```
idevicebackup2 backup --full iphone_backup_folder
```

jokerekv2, 2021-08-23:
> did you do full backup with idevicebackup?
> 
> itunes explictily call full backup, if there is no previous backup present (or if there is big time difference from old backup)
> 
> idevicebackup calls service with "default" values ... what is default knows only apple, but if you want to make sure to do "full" backup you need to call with -- full parameter
> 
> ```
> idevicebackup2 backup --full iphone_backup_folder
> ```
> 
> as what is actually backed up, it's up to device. backup is controlled by device, not by application in computer (that is true for both, itunes and idevicebackup), also backup contains more data, if it's password protected.

I did a full backup and set encryption to on. The backup is always 4Gb. It says in the console that a full backup is done. I am worried that when I want to go back to this backup there will be a problem....
Btw i installed libimobiledevice in `/opt/local/` as in the tutorial. Now I run the commands using:
`/opt/local/bin/idevicebackup2`.
Is everything ok? I noticed that I cannot run the **list** option for idevicebackup2.

nikias, 2021-08-23:
The `--list` option is broken, since Apple change the backup structure and basically abandoned that feature. As for the iTunes backup, have you checked if it has some `MobileSync/Backup/<UDID>/Snapshot` data? Also compare the sizes of the `Manifest.db` files for both backups.

mexmer, 2021-08-23:
list option doesn't work since apple changed database format, which was in ios10 i think (changed from mdbd to sqlite)

if you want to compare backup contents, there are iphone backup browsing tools, some free, some paid.

app i'm developing using code based on libimobile for making apple backups (and restoring them), and we never observed any data loss. it's possible that itunes might retrieve some additional data.

also make sure, there is no "Snapshot" folder inside backup directory, which might contain copy of backup from unsuccesfull run.

jokerekv2, 2021-08-23:
There is no Snapshot folder. Ok then.. now im trying to unback to see what's there. But im getting error like this:
```
ERROR: Could not receive from mobilebackup2 (-256)
Unback Aborted.
```

jokerekv2, 2021-08-23:
Next question.. i install it again but in not /opt/ directory.. im getting this error:
```
minis@minis-zorin:~$ idevicebackup2 backup --full /home/minis/Documents/tmp
Backup directory is "/home/minis/Documents/tmp"
Started "com.apple.mobilebackup2" service on port 56817.
Could not perform backup protocol version exchange, error code -1
```
Installed it using: 
```
./autogen.sh
make
sudo make install
```
Version is 1.3.1 so installation i think was okey. Before was 1.3.0...

nikias, 2021-08-23:
unback is just as broken as list...

nikias, 2021-08-23:
It's even worse than I thought. Turns out, `list` and `info` cannot work with an encrypted backup, because Apple's code on the device doesn't forward the backup password (which idevicebackup2 doesn't pass, but I changed that to test it), and errors out with an error
```
ErrorCode 207: Invalid password when restoring encrypted backup (MBErrorDomain/207). Underlying error: MKBKeyBagUnlock error (-3) (MBErrorDomain/207).
```
I guess I need to print a warning when either one of the `info`, `list`, or `unback` commands is used...

jokerekv2, 2021-08-23:
Understand.. so at this moment it is unusable. But what about libimobiledevice installation? What could be the problem of installing this way:
```
./autogen.sh
make
sudo make install
```
Why it isn't working?
And why this way is working?
```
./autogen.sh --prefix=/opt/local --enable-debug
make
sudo make install
```

nikias, 2021-08-23:
I assume you have another older installation in `/usr/bin` and your own build in`/usr/local/bin` which is the default prefix for most platforms). In your case the path `/opt/local/bin` comes before `/usr/bin` but `/usr/local/bin` after. Check your `PATH` environment variable.
You can run `which idevicebackup2` (or any other command) so see which path is being used.

jokerekv2, 2021-08-23:
```
minis@minis-zorin:~$ which idevicebackup2 
/usr/local/bin/idevicebackup2
minis@minis-zorin:~$ idevicebackup2 --version
idevicebackup2 1.3.1
minis@minis-zorin:~$ 
```
It's using the right one I think. Before installation there was 1.3.0 version.

mexmer, 2021-08-24:
@jokerekv2 uninstall your linux distribution provided package of libimobiledevice, libplist, libusbmuxd, usbmuxd and use only version compiled from sourcecode

Boss-Chicken, 2022-01-08:
I have this same issue as well.

Whilst I haven't done an iTunes backup to compare with the Linux one, I do know that an iPhone that's used ~23GB that has a backup that's _also encrypted_ shouldn't be ~11GB in size.

I've noticed mentions of downloading from source in this thread and some others, considering that my version from my distribution seems to be decently old (`idevicebackup2 --version` shows my version as `idevicebackup2 1.3.0`!).

I'll try to compile/download from source or whatever it's called and update this information when it happens.

Also want to ask if it will save iMessages AND the photos/videos from it.

Boss-Chicken, 2022-01-08:
I've just tried removing the packages @mexmer recommended to remove and had some conflicting dependencies.

`sudo pacman -R libimobiledevice libplist libusbmuxd usbmuxd`
output
`checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing libimobiledevice breaks dependency 'libimobiledevice' required by gvfs-afc
:: removing usbmuxd breaks dependency 'usbmuxd' required by gvfs-afc
:: removing libimobiledevice breaks dependency 'libimobiledevice' required by upower`

When I tried doing `sudo pacman -U libimobiledevice libplist libusbmuxd usbmuxd` the terminal showed that I somehow _did_ uninstall it. It might've been something that I didn't notice from the error earlier as I'm still very new into Linux and getting comfortable with the command line.

The output for the command `sudo pacman -U libimobiledevice libplist libusbmuxd usbmuxd` was this:

`loading packages...
error: 'libimobiledevice': could not find or read package
error: 'libplist': could not find or read package
error: 'libusbmuxd': could not find or read package
error: 'usbmuxd': could not find or read package`

I installed it again, and did the same upgrade command again, the same result is shown by the terminal.

The output for the command `idevicebackup2 --version` and `which idevicebackup2` shows these results respectively:

`idevicebackup2 1.3.0`

`/usr/bin/idevicebackup2`

Yet anything to do with libimobiledevice seems to show that it doesn't exist or isn't installed.

I'll try once again to install/compile from source with a git clone or however one does it, just wanted to put this reply up to keep track of what could be wrong with the versions or catch myself in the act of being a noob etc.

mexmer, 2022-01-10:
don't install any distribution packages, only remove them, and then download source and compile it from source
and yes, some other applications might be depending on it (gvfs-afc is gvfs plugin for accessing iphone camera files, after connecting to usb i think)

windowalter, 2023-11-07:
idevicebackup2 always wants to reboot device for complete restore.Can we do it without reboot

brandon-fryslie, 2024-07-08:
Figuring out which binaries/executables and libraries you are using and compiling software from source is not really related to libimobiledevice.  Dependency issues (and without) are just a normal part of using Linux (no value judgement, eventually you learn it or quit Linux I reckon).  I would highly recommend you learn and use Docker for situations like this, rather than using computer's file system to experiment with unstable software.  You can do all of this in a way thats reproducible, self-documenting, and doesn't leave a mess on your computers file system using Docker.  

As a bonus, you can just copy from someone else who's already gotten the work started for you:
- https://github.com/ymin/libimobiledevice-docker
- https://hub.docker.com/r/lukepasek/idevicerestore

This discussion has veered way off topic and this issue can probably be closed.