libimobiledevice/libimobiledevice #1207: ideviceinfo doesn't detect iPhone 5

Midou36O, 2021-08-29:
when plugging the iPhone it indicates it is charging then it goes back to non-charging state (this isn't a cable issue as on windows it works perfectly fine).

looking at the dmesg output the iPhone seems to be detected in linux: (DFU/Recovery modes are also detected, and for some reason these works fine on linux and gets recognized.)

```
[ 8104.727716] usb 1-3: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice= 5.20
[ 8104.727723] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8104.727726] usb 1-3: Product: iPhone
[ 8104.727729] usb 1-3: Manufacturer: Apple Inc.
[ 8104.727732] usb 1-3: SerialNumber: 4b9d2d77b73085e*****32620f3666d42ebf404
[11178.192238] apple-mfi-fastcharge 1-3: USB disconnect, device number 6
[11179.895266] usb 1-3: new high-speed USB device number 7 using ehci-pci
[11180.048111] usb 1-3: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice= 5.20
[11180.048120] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11180.048124] usb 1-3: Product: iPhone
[11180.048126] usb 1-3: Manufacturer: Apple Inc.
[11180.048129] usb 1-3: SerialNumber: 4b9d2d77b73085e******32620f3666d42ebf404
```
but idevice doesn't find anything

```
❯ sudo ideviceinfo
ERROR: No device found!
```
ifuse doesn't work too:
```
❯ sudo ifuse /run/iPhone
ERROR: No device found!
Is the device properly connected?
If it is make sure that your user has permissions to access the raw USB device.
If you're still having issues try unplugging the device and reconnecting it.
```

OS: Artix Linux Open-RC (Arch Linux without systemd)
iPhone model: iPhone 5 Global iOS 10.3.3

mexmer, 2021-08-30:
ideviceinfo needs usbmuxd. do you have usbmuxd istallled and running?

Midou36O, 2021-08-30:
Yes i installed usbmuxd and it is running.

mexmer, 2021-09-01:
perhaps, shutdown usbmuxd, run it as root on foreground and logging, and replug device again, we might see what is going on,
```
usbmuxd -fvv
```
if your system is using udev, also add u switch eg.
```
usbmuxd -fvvu
```

Midou36O, 2021-09-01:
output of usbmuxd :

Midou36O, 2021-09-01:
crap i closed the issue

nikias, 2021-09-01:
The latest output shows the device being handled fine and preflight succeeding. Does ideviceinfo still not find it?

nikias, 2021-09-05:
Well the problem is when you run in manually you should NOT pass the `-u` command line switch, because then this happens:
`[15:01:18.042][3] Automatic device discovery on hotplug disabled.`
The way it works with udev is that when launching another instance it would trigger device discovery, but this requires a proper udev rule installed (and udev working).

Midou36O, 2021-09-06:
The same things happens if i omit `-u`  command line switch:
device fails to get detected when plugging it **after** launching `usbmuxd -fvv`
device gets recognized when plugging it **before** launching `usbmuxd -fvv`

i can get used to manually trigger usmuxd if there is no solution to detecting the device after launching usbmuxd :P

closed 2021-09-06 by Midou36O

nikias, 2021-09-06:
Since it works fine in multiple Linux environments it must be something wrong with your setup. Maybe it's libusb related or maybe it's something else. For example it could be an older libimobiledevice which then results in preflight errors, and then the device won't be visible. How does the output look like if you run `sudo usbmuxd -fvv` and then when it runs, plug in a device?

Midou36O, 2021-09-07:
[output.txt](https://github.com/libimobiledevice/libimobiledevice/files/7121173/output.txt)
[output2.txt](https://github.com/libimobiledevice/libimobiledevice/files/7121176/output2.txt)

nikias, 2021-09-07:
Output looks good, does ideviceinfo still not work? I am seeing `upower` and `iphone-set-info`, if those are using some really old libimobiledevice they could break the device pairings though, so maybe those are making things fail for you. But the device detection works fine.

Midou36O, 2021-09-07:
First time tend to fail but after rerunning usbmuxd for a second time it works perfectly fine. I'm pretty sure that i'm running the latest git version (i'm building from sources) 🤔

Silfalion, 2021-09-23:
Have the same issue as well

chamychu, 2023-11-22:
just got the same issue on arch with my iphone 13 pro max

Keylessboi, 2024-07-05:
Why was this closed if others are having the same issue, which by the way I'm having the same issue myself on arch