libimobiledevice/libimobiledevice #1208: idevicedebug not working (iOS 14.6 iPhone XR)

Abelincoln09, 2021-08-31:
No matter what I do, I mount the image and I run idevicedebug but regardless of what I do or how I place the command for what I'm doing it just displays the usage text for the command.
If it helps, my command line is idevicedebug -u <UDID> -detach run <APP ID>
(Trying to run the command for dolphin ios)

Abelincoln09, 2021-08-31:
Nevermind, it suddenly decided to work

closed 2021-08-31 by Abelincoln09

mexmer, 2021-09-01:
if you have issue running debug after mounting dev image restart your phone and do it again, there might be already different image loaded in phone, or just not enough memory to process mounted image (which will silently fail, since command are run inside phone not giving output). 

i've had this issue even with xcode, when i was not able to debug application, until i restarted phone ... xcode didn't produce any error, except cannot connect to debug channel and cannot start debug.