libimobiledevice/libimobiledevice #1209: [idevicediagnostics] WiFi - Active key's value is always "NO"

pespinel, 2021-09-02:
The WiFi Active key value is always "NO":
```
[#MacBookPro ~ ]$idevicediagnostics diagnostics WiFi
```
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>WiFi</key>
	<dict>
		<key>Active</key>
		<string>NO</string>
		<key>Status</key>
		<string>Success</string>
	</dict>
</dict>
</plist>

mexmer, 2021-09-02:
those values come from device, so if apple sends NO, then there is nothing we can do about it.

it's not first value, that apple simply stopped updating, most likely there is new value somewhere else, but that only apple knows.

pespinel, 2021-09-03:
Ok, @mexmer thanks for the explanation. Do you know any method to retrieve the WiFi status via terminal?

mexmer, 2021-09-07:
sorry, if i knew i would already wrote it here, but unfortunately i don't.

pespinel, 2021-09-13:
Probably what I'm going to say doesn't make sense but should We be able to find the new key easily in a jailbroken device?

mexmer, 2021-09-13:
maybe, they might have mirrored this value in mobilegestalt output, but you will need to find, which value is that.

you can query mobilegestalt keys with idevicediagnostics, or write own app, that will run on jailbroken phone
keys can be queried with hash or with name ... there is incomplete list for ios12
https://blog.timac.org/2018/1126-deobfuscated-libmobilegestalt-keys-ios-12/

no idea if it will work with 14 or 15, but you can try.

pespinel, 2021-09-15:
I've created a simple script to call idevicediagnostics for all the known keys of iOS 12 and make a diff between the iPhone with the wifi ON and with the wifi OFF and, unfortunately, none of them was  the right one :(

I'll keep searching ...