libimobiledevice/libimobiledevice #1212: How to connect iOS device with wifi under win10?

HackHarry, 2021-09-06:


lostwindsenril, 2021-10-08:
I think wifi connection works well without additional steps if you have Apple Drivers installed well. Isn't it?
[image]

HackHarry, 2021-10-21:
> I think wifi connection works well without additional steps if you have Apple Drivers installed well. Isn't it? [image]

Thanks for your reply. But in my win10 PC, I only can get USB devices, and my wifi is hotspot without proxy. Maybe some drivers needed?
[image]