libimobiledevice/libimobiledevice #1216: AppleTV 4K visible over network from macOS but not from Debian virtual machine on same laptop

JasonLunn, 2021-09-20:
I'm working to build a Debian-based system that can interrogate an AppleTV 4K. AppleTV 4K's have no USB port, so I have no choice but to rely on the network-based connection support.

On my MacBook Pro, `idevice_id -n` prints the UDID of the AppleTV. The AppleTV in question is literally new out of the box - it has been plugged into power, network, and a TV, and literally nothing else. I haven't even touched the remote.

On the same MacBook Pro, I have a Debian virtual machine running inside VirtualBox. `idevice_id -n` prints nothing. I can confirm that I can ping the AppleTV's IP from virtual machine, so I'm reasonably confident that there is no network connectivity issue.

Is this a limitation of libimobiledevice's `usbmuxd`'s ability to mimic the native `usbmuxd` on macOS? Is this a regression introduced between 1.30 (the version installed via homebrew on macOS) and 1.3.1 (the version I built from source)? Was there a feature flag I failed to include to make this work?

Environment:
  MacBook Pro:
   CPU: `2.6 GHz 6-Core Intel Core i7`
   OS: `Big Sur 11.6 (20G165)`
   idevice_id version: `1.3.0` (via homebrew)
   VirtualBox: `6.1.26 r145957 (Qt5.6.3)`
 Virtual machine:
   OS: `Linux debian 5.10.0.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux`
   idevice_id version: `1.3.1` (via built from source downloaded from Github 2021-09-20)
 AppleTV 4K:
   DeviceName: `Apple TV`
   HardwareModel: `J105aAP`
   ProductName: `Apple TVOS`
   ProductType: `AppleTV6,2`
   ProductVersion: `11.1`
   Network: Wired Ethernet

mexmer, 2021-09-21:
support for network connected apple devices is in progress, but has not been finished yet. it's not regression, because it never worked, atm. usbmuxd supports only usb connected devices.

if you run ideviceinfo or idevice_id under OS X, it doesn't connect to usbmuxd (in fact usbmuxd is not running at all), it uses apple service, that is responsible for connecting apple device and it indeed does see iDevices that are on same network as OS X computer (either wired or wireless).

mexmer, 2021-09-22:
i think there is fork of usbmuxd, but main issue is, that while on mac their service uses bonjour protocol for service discovery, there is no such thing on linux.
i mean there exists compatible implementation (avahi), but i have no idea how easy or hard it will be to integrate avahi support to usbmuxd. that might be question to @nikias . not to mention, i have no idea, what is general availability of avahi trough linux distributions.

scj643, 2021-09-22:
https://github.com/tihmstar/usbmuxd2 is the fork you're probably thinking of. I managed to get it compiled in RHEL and got wireless pairing working but not for Apple TV.

mexmer, 2021-09-23:
that's look like it. seems it's uses avahi-client for searching wifi devices.

bentumbler, 2021-10-01:
I documented the full set of steps to get the tihmstar re-implementation working on a Raspberry Pi. https://www.numerousnetworks.co.uk/guides/nmirror/
 (and it includes a BT PAN connection for anyone interested)