libimobiledevice/libimobiledevice #1217: Homebrew installation of libimobiledevice from HEAD fails with missing package libimobiledevice-glue-1.0

ewscripps-jb, 2021-09-21:
Trying to install libimobiledevice from HEAD on macOS Big Sur version 11.6 fails:

164485@EWS2000254 ~ % brew install --HEAD libimobiledevice
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/164485/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at bd2e446 [github-actions] Add an explicit apt-get update before attempting to install packages on ubuntu
==> ./autogen.sh
Last 15 lines from /Users/164485/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh:
checking for pkg-config... /usr/local/Homebrew/Library/Homebrew/shims/mac/super/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd-2.0 >= 2.0.2... yes
checking for libplist-2.0 >= 2.2.0... yes
checking for libimobiledevice-glue-1.0 >= 1.0.0... no
configure: error: Package requirements (libimobiledevice-glue-1.0 >= 1.0.0) were not met:

No package 'libimobiledevice-glue-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables limd_glue_CFLAGS
and limd_glue_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

READ THIS: https://docs.brew.sh/Troubleshooting

Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels.

felipemengatto, 2021-09-22:
Does anyone have any idea how to solve this?

mexmer, 2021-09-22:
you need to install libimobiledevice-glue ... no idea if it's available on homebrew, but it is available on github
https://github.com/libimobiledevice/libimobiledevice-glue

nikias, 2021-09-22:
The brew formulae are outdated, and there isn't one for libimobiledevice-glue yet.

felipemengatto, 2021-09-22:
In our case, we cloned the project and did the build/install manually. For now let's keep this workaround.
Thanks @edugaedke for your help.

[image]

ewscripps-jb, 2021-09-22:
So should Homebrew be used to install the dependencies (libusbmuxd, libplist, etc.) or should each project be downloaded, then build and install?  Will a combination of brew installs and local builds possibly cause confusion?

 ~ % brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/libimobiledevice-glue-1.0.0.dylib

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/libimobiledevice-glue/collection.h
  /usr/local/include/libimobiledevice-glue/socket.h
  /usr/local/include/libimobiledevice-glue/termcolors.h
  /usr/local/include/libimobiledevice-glue/thread.h
  /usr/local/include/libimobiledevice-glue/utils.h

Warning: Unbrewed '.la' files were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected '.la' files:
  /usr/local/lib/libimobiledevice-glue-1.0.la

Warning: Unbrewed '.pc' files were found in /usr/local/lib/pkgconfig.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected '.pc' files:
  /usr/local/lib/pkgconfig/libimobiledevice-glue-1.0.pc

Warning: Unbrewed static libraries were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected static libraries:
  /usr/local/lib/libimobiledevice-glue-1.0.a

ewscripps-jb, 2021-09-29:
@felipemengatto How did you get libplist to see that libimobiledevice-glue-1.0 was installed?  Is there another step after make and make install?

felipemengatto, 2021-10-04:
There is no longer any command, we just installed other libraries that we used and everything worked fine.

```.yml
          - brew install libplist || echo 'fails because libimobiledevice-glue is missing'
          - git clone https://github.com/libimobiledevice/libimobiledevice-glue.git
          - cd libimobiledevice-glue
          - ./autogen.sh
          - make
          - sudo make install
          - cd ..
          - brew install librsvg
          - brew install imagemagick
          - brew upgrade node
          - cd ios
          - RBENV_VERSION=2.5.1 pod install
```

ewscripps-jb, 2021-10-06:
@felipemengatto - Thank you for your time!  I will give it another try to see if I can get libimobiledevice --HEAD installed on macOS Big Sur.

topeerz, 2022-01-10:
I was able to make it work on my machine ... There seem to be two issues to be addressed;
**1) The installation of libimobiledevice-glue.**
There seem to be no brew formula for that yet.
This may be addressed by downloading & building library separately of brew OR by creating new brew rule via
> brew create "https://github.com/libimobiledevice/libimobiledevice-glue.git"
> brew edit libimobiledevice-glue
> brew install --HEAD libimobiledevice-glue

My crude rule looks like this (it should be optimised and I removed inline comments here hence weird line numbers).
``` 
  4 class LibimobiledeviceGlue < Formula
  5   desc ""
  6   homepage ""
  7   license ""
  8   head "https://github.com/libimobiledevice/libimobiledevice-glue.git"
 11
 12   depends_on "autoconf" => :build
 13   depends_on "automake" => :build
 14   depends_on "libtool" => :build
 15   depends_on "pkg-config" => :build
 16   depends_on "libplist"
 17
 18   def install
 22     system "./autogen.sh", "--prefix=#{prefix}"
 30     system "make", "install"
 31   end
 32
 33 end
```

**2 Installation of libimobiledevice head**
In order to do that I have updated original recipe by adding reference to `libimobiledevice-glue` I have just installed as dependance. The brew seems to create custom `PKG_CONFIG_PATH` which overrides the system one so without this step I was seeing pkg tool errors (as described in previous comments).
> brew edit libimobiledevice

```
 31   depends_on "libimobiledevice-glue"
```

At this point I was able to run successfully (please note you may need to unlink previous version of libimobiledevice  if it has been already installed)
> brew install --HEAD libimobiledevice

**Debugging**
I was able to get some details by running 
> brew install --HEAD libimobiledevice --verbose --debug
Once build failed I have select 5 (shell). Then I was able to call
> pkg-config libimobiledevice-glue-1.0 --modversion
to verify lib is found as well as version visible to brew ...

tomriddly, 2022-01-10:
已收到您发的邮件。

tomriddly, 2022-01-11:
> 
This is a much easier way approved to be useful.

PS:
`depends_on "libimobiledevice-glue"`
libusbmuxd will also need this line.

annazolkieve, 2022-01-28:
install:
```
$ make install
Making install in src
 /opt/local/bin/gmkdir -p '/usr/local/lib'
 /bin/sh ../libtool   --mode=install /opt/local/bin/ginstall -c   libimobiledevice-glue-1.0.la '/usr/local/lib'
libtool: install: /opt/local/bin/ginstall -c .libs/libimobiledevice-glue-1.0.0.dylib /usr/local/lib/libimobiledevice-glue-1.0.0.dylib
libtool: install: (cd /usr/local/lib && { ln -s -f libimobiledevice-glue-1.0.0.dylib libimobiledevice-glue-1.0.dylib || { rm -f libimobiledevice-glue-1.0.dylib && ln -s libimobiledevice-glue-1.0.0.dylib libimobiledevice-glue-1.0.dylib; }; })
libtool: install: /opt/local/bin/ginstall -c .libs/libimobiledevice-glue-1.0.lai /usr/local/lib/libimobiledevice-glue-1.0.la
libtool: install: /opt/local/bin/ginstall -c .libs/libimobiledevice-glue-1.0.a /usr/local/lib/libimobiledevice-glue-1.0.a
libtool: install: chmod 644 /usr/local/lib/libimobiledevice-glue-1.0.a
libtool: install: ranlib /usr/local/lib/libimobiledevice-glue-1.0.a
 /opt/local/bin/gmkdir -p '/usr/local/lib/pkgconfig'
 /opt/local/bin/ginstall -c -m 644 libimobiledevice-glue-1.0.pc '/usr/local/lib/pkgconfig'
Making install in include
make[2]: Nothing to be done for `install-exec-am'.
 /opt/local/bin/gmkdir -p '/usr/local/include'
 /opt/local/bin/gmkdir -p '/usr/local/include/libimobiledevice-glue'
 /opt/local/bin/ginstall -c -m 644  libimobiledevice-glue/socket.h libimobiledevice-glue/thread.h libimobiledevice-glue/utils.h libimobiledevice-glue/collection.h libimobiledevice-glue/termcolors.h '/usr/local/include/libimobiledevice-glue'
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
```
then:
```
configure: error: Package requirements (libimobiledevice-glue-1.0 >= 1.0.0) were not met:

No package 'libimobiledevice-glue-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables limd_glue_CFLAGS
and limd_glue_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```
why? how to fix?

tomriddly, 2022-01-28:
已收到您发的邮件。

annazolkieve, 2022-01-28:
@annazolkieve fixed with `export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig/`

KostyaSha, 2023-07-14:
> brew edit libimobiledevice

jfyi brew needs env var either it will switch to original file instead
https://docs.brew.sh/FAQ#can-i-edit-formulae-myself

dnicolson, 2023-11-23:
There is a formula now:
```
brew install libimobiledevice-glue
```

jpstotz, 2024-08-19:
And now the missing formula for `libtatsu` prevents installation of head version of libimobiledevice.

tux-mind, 2024-11-06:
I'm trying to fix it by [contributing](https://github.com/Homebrew/homebrew-core/pull/196866) to homebrew. However, `libtatsu` doesn't meet some requirements for the Formulae:
```
* GitHub repository not notable enough (<30 forks, <30 watchers and <75 stars)
```

Can you please fork and star the [project](https://github.com/libimobiledevice/libtatsu) to reach those numbers? That would speed up the PR merging I guess.

Thank you 🙏