libimobiledevice/libimobiledevice #1219: Mac OSX M1 complie failed

badashanren, 2021-09-24:
Undefined symbols for architecture arm64:
  "_ASN1_INTEGER_free", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_INTEGER_new", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_INTEGER_set", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_TIME_free", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_TIME_new", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)

dnicolson, 2023-11-23:
I was also seeing these errors, including these warnings:
```
ld: warning: ignoring file '/usr/local/Cellar/openssl@3/3.1.1_1/lib/libcrypto.3.dylib': found architecture 'x86_64', required architecture 'arm64'
ld: warning: ignoring file '/usr/local/Cellar/openssl@3/3.1.1_1/lib/libssl.3.dylib': found architecture 'x86_64', required architecture 'arm64'
```

The solution is to remove the `x86_64` formula:
```
rm -rf /usr/local/Cellar/openssl@3
```

The formula was probably installed when using `arch -x86_64` commands as part of Apple's 
Game Porting Toolkit.

mexmer, 2023-11-24:
@dnicolson on arm based mac (eg. with apple silicon cpu), you can use either x86_x64 or arm64 arch, but all parts of project need to be compiled for same architecture, and that also included dependent libraries. 
so there are multiple options.
1) reinstall dependent library with architecture matching your project configuration
2) reconfigure and rebuild your project with architecture matching installed libraries
3) setup your enviroment with multiple archicture, and switch to proper architecture when building project (this one is most complex, but you can find guides on internet, how to do that)

i personaly use arm64 architecture on mac silicon, and if required library is not in homebrew for arm64, i just build it from source.

ruoyush, 2024-08-02:
> I was also seeing these errors, including these warnings:
>
> ```
> ld: warning: ignoring file '/usr/local/Cellar/openssl@3/3.1.1_1/lib/libcrypto.3.dylib': found architecture 'x86_64', required [...]

I found this post when searching for a solution to my problem.. I recently ran into a similar issue. In my case, I have one same library under x86 and arm homebrew. When I tried to compile my project, it always looks for the x86 path /usr/local instead of /opt/homebrew. My solution is installing pkg-config in arm homebrew, so pkg_search_module() in my CMake file will search for the libraries under /opt/homebrew.