libimobiledevice/libimobiledevice #1220: Device Restore issue on iOS 15 Devices

kalit-zz, 2021-09-27:
I am facing an issue to restore devices that have iOS 15 as OS

"Error": EnterRecoveryCommand - 656756374519854 00008020-000255510268402E 
Failed to enter recovery mode.

We faced same kind of issue in March this year, the solution is provided in this commit - https://github.com/libimobiledevice/libimobiledevice/commit/25059d4c7d75e03aab516af2929d7c6e6d4c17de

kalit-zz, 2021-09-28:
We are to restore all the iOS under iOS 15 version but facing issues on iOS 15 Devices

nikias, 2021-09-28:
The requirements to enter recovery mode didn't change in iOS 15 and so that commit should have fixed it for you. It definitely works, I just tested it.

kalit-zz, 2021-09-28:
As per our latest findings, the recovery is working fine but restoration
part is failing for iOS 15. Same work flow is working fine for iOS 14 and
below. The devices are strucking on the restoration mode.

I'll be thankful to you if you can point me in right direction.

nikias, 2021-10-01:
Update https://github.com/libimobiledevice/libirecovery.