libimobiledevice/libimobiledevice #1222: Error writing keychain backup at path

qbarnes, 2021-09-28:
I ran:
```
$ sudo idevicebackup2 backup  ~/iphone-backup
```
It ran for some time getting pretty far then failed with the error:
```
[==================================================] 100% (12.7 MB/11.6 MB)     
[================================================= ]  98% Finished
ErrorCode 100: Error writing keychain backup at path "/var/Keychains/keychain-backup.plist.tmp" (MBErrorDomain/100). Underlying error: The operation couldn’t be completed. (com.apple.security.xpc error 3 - <connection: 0x283dac820> { name = com.apple.securityd, listener = false, pid = 0, euid = 4294967295, egid = 4294967295, asid = 4294967295 }: Connection interrupted) (com.apple.security.xpc/3).
Received 106084 files from device.
Backup Failed (Error Code 100).
```

I built and installed `libimobiledevice-glue` and `libimobiledevice` from their master branches on Fedora 34.  I'm running iOS 15.0 using an iPhone 11 Pro.

This is the first time I've attempted to run this software, so I have no basis for comparison.

I tried googling for the problem, but only found the error message on a few Chinese web sites that I can't read and they didn't successfully translate.

Any suggestions for things to try or check?

nikias, 2021-09-28:
This error happens on the device. Seems like you are trying an unencrypted backup, I wonder if it works when you turn on backup encryption?
```sh
idevicebackup2 -i encryption on
```

qbarnes, 2021-09-28:
I ran it and got:
```
$ sudo idevicebackup2 -i encryption on
Started "com.apple.mobilebackup2" service on port 55853.
Negotiated Protocol Version 2.1
ERROR: Backup encryption is already enabled. Aborting.
```

mexmer, 2021-09-29:
first thing you should do is restart phone, might be just not enough memory in device ... while iOS has amazing memory management, some operations (and backup is one of them), require lot of memory, and if they throw weird errors, like this ... restart might help.

also check free space on device, although if there was issue with free space on device, it would probly say so, with different error.  from my experience, making itunes backup on devices with less than 200MB of free space is almost impossible, backup will simply not be able to create all tempfiles it requires)

qbarnes, 2021-09-29:
> ... restart might help.

That did the trick.

Also, when backing up with `,idevicebackup2` before I had kept an app running.  This time I let the phone sit on its home screen and sleep.  I had left the app running because it's one that keeps the screen active and doesn't let the phone go to sleep.  I have an old Mac running El Capitan and with its iTunes, if the phone sleeps while backing up, it would often hang the backup.  The app was a workaround for that problem.  Looks like I don't have to worry about that with `idevicebackup2`.

> also check free space on device

Not a problem.  I have more than 100GB of free storage.

Thanks!

closed 2021-09-29 by qbarnes

mexmer, 2021-09-29:
> 
> > also check free space on device
> 
> Not a problem. I have more than 100GB of free storage.
> 
> Thanks!

i did mean free space inside phone, not your mac (although if you have phone with 256GB or more, you can have that much free, my 128GB ipad and 128 SE 2020 have both less than 30GB free 😸 )

qbarnes, 2021-09-29:
Yes, I have a 256GB iPhone and am using a little over half its storage.

paulschreiber, 2022-02-01:
Restarting the iPhone solved the problem for me as well (macOS 12.2, iOS 15.3).