libimobiledevice/libimobiledevice #1224: idevicebackup2 - commands info & list with error

ghollf, 2021-10-03:
```
$ idevicebackup2 -d info .
ERROR: Could not receive from mobilebackup2 (-256)
$ idevicebackup2 -d list .
ErrorCode 205: Error reading backup database (MBErrorDomain/205). Underlying error: No such file or directory at path "547d44f60b4d9bd654b3561d3ccd7dd76113fbfd/Manifest.mbdb" (MBErrorDomain/4).
user1@papa:/media/user1/EXT$ idevicebackup2 -v
$ ls 547d44f60b4d9bd654b3561d3ccd7dd76113fbfd
00  0b  16  21  2c  37  42  4d  58  63  6e  79  84  8f  9a  a5  b0  bb  c6  d1  dc  e7  f2  fd
01  0c  17  22  2d  38  43  4e  59  64  6f  7a  85  90  9b  a6  b1  bc  c7  d2  dd  e8  f3  fe
02  0d  18  23  2e  39  44  4f  5a  65  70  7b  86  91  9c  a7  b2  bd  c8  d3  de  e9  f4  ff
03  0e  19  24  2f  3a  45  50  5b  66  71  7c  87  92  9d  a8  b3  be  c9  d4  df  ea  f5  Info.plist
04  0f  1a  25  30  3b  46  51  5c  67  72  7d  88  93  9e  a9  b4  bf  ca  d5  e0  eb  f6  Manifest.db
05  10  1b  26  31  3c  47  52  5d  68  73  7e  89  94  9f  aa  b5  c0  cb  d6  e1  ec  f7  Manifest.plist
06  11  1c  27  32  3d  48  53  5e  69  74  7f  8a  95  a0  ab  b6  c1  cc  d7  e2  ed  f8  Status.plist
07  12  1d  28  33  3e  49  54  5f  6a  75  80  8b  96  a1  ac  b7  c2  cd  d8  e3  ee  f9
08  13  1e  29  34  3f  4a  55  60  6b  76  81  8c  97  a2  ad  b8  c3  ce  d9  e4  ef  fa
09  14  1f  2a  35  40  4b  56  61  6c  77  82  8d  98  a3  ae  b9  c4  cf  da  e5  f0  fb
0a  15  20  2b  36  41  4c  57  62  6d  78  83  8e  99  a4  af  ba  c5  d0  db  e6  f1  fc
```

ghollf, 2021-10-03:
#1058 seems to explain the issue already. The feedback of the commands could possibly be improved.

mexmer, 2021-10-04:
it's ios 10. so info command is broken ... apple broke it, and didn't fix it even in ios 15. as stated in issue you linked.

closed 2022-03-12 by ghollf