libimobiledevice/libimobiledevice #1240: Unable to backup

onesign12, 2021-11-17:
`idevicebackup2 backup ~/Documents/iPhoneBackups/
Backup directory is "/home/eleontev/Documents/iPhoneBackups/"
Started "com.apple.mobilebackup2" service on port 55029.
Could not perform backup protocol version exchange, error code -1`

Ubuntu 20.04 LTS
`ideviceinfo -s
BasebandCertId: ----
BasebandKeyHashInformation: 
 AKeyStatus: 64
 SKeyStatus: 2
BasebandSerialNumber: ---
BasebandVersion: 3.04.01
BoardId: 14
BuildVersion: 18G82
CPUArchitecture: arm64e
ChipID: 32800
DeviceClass: iPhone
DeviceColor: 1
DeviceName: iPhone
DieID: 3435107162521658
HardwareModel: D321AP
HasSiDP: true
PartitionType: GUID_partition_scheme
ProductName: iPhone OS
ProductType: iPhone11,2
ProductVersion: 14.7.1
ProductionSOC: true
ProtocolVersion: 2
SupportedDeviceFamilies[1]: 
 0: 1
TelephonyCapability: true
UniqueChipID: ---
UniqueDeviceID: ---
WiFiAddress: ---
`

mexmer, 2021-11-18:
you need to use latest version of idevicebackup and libimobiledevice, this has been already said multiple times.

#1112 , #1111, #1032, #1028, #1013, #1001, #1115

FYI github issue tracker has search function

dickorydock, 2022-03-07:
Hi, did you ever find a way to fix this? I notice the above comment says you need to install latest versions but it does not say how to install those versions. Thank you.

lestephane, 2022-03-20:
Tried using Ubuntu 21.10 (which has the latest versions) in a Vagrant VM, same problem.

## Versions
```
$ dpkg -l  "libimobiledevice*" "libusbmuxd*" "libplist*" | grep -vE "^un"
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                    Version       Architecture Description
+++-=======================-=============-============-==================================================================
ii  libimobiledevice-utils  1.3.0-6       amd64        Utitilies for communicating with iPhone and other Apple devices
ii  libimobiledevice6:amd64 1.3.0-6       amd64        Library for communicating with iPhone and other Apple devices
ii  libplist3:amd64         2.2.0-6build1 amd64        Library for handling Apple binary and XML property lists
ii  libusbmuxd6:amd64       2.0.2-3build1 amd64        USB multiplexor daemon for iPhone and iPod Touch devices - library
```

## Vagrantfile
```
cat <<EOF > Vagrantfile
Vagrant.configure("2") do |config|
  config.vm.box = "ubuntu/impish64"
  config.vm.provider "virtualbox" do |vb|
    vb.customize ["modifyvm", :id, "--usb", "on"]
    vb.customize ["modifyvm", :id, "--usbehci", "on"]
  end
  # doc: https://docs.oracle.com/en/virtualization/virtualbox/6.0/user/vboxmanage-usbfilter.html
  config.vm.provider "virtualbox" do |vb|
    vb.customize ["usbfilter", "add", "0",
        "--target", :id,
        "--name", "Any iPhone",
        "--product", "iPhone"]
  end
  config.vm.provision "shell", inline: <<-SHELL
    apt-get update
    apt-get install -y usbmuxd libimobiledevice6 libimobiledevice-utils
  SHELL
end
```
- Then within the VM, run the backup command
```
$ vagrant ssh
$ idevicebackup2 backup --full IphoneBackup --debug --interactive
...
Backup directory is "IphoneBackup"
Started "com.apple.mobilebackup2" service on port 52096.
Could not perform backup protocol version exchange, error code -1
```