libimobiledevice/libimobiledevice #1241: lockdownd_client_free block

xanadu-lee, 2021-11-18:
I connected a device and connected lockdown through lockdownd_client_new_with_handshake.
Now that the device is disconnected, I release memory by calling lockdownd_client_free.
However, it will get stuck inside lockdownd_client_free and return for a long time (more than 10 seconds). Not every time.
What's the problem? How to solve it.

closed 2023-01-07 by xanadu-lee

AiXanadu, 2023-08-17:
Does anyone know?

mexmer, 2023-08-17:
if you look at the code, you can see the reason. it's because it tries to do it gracefully, if lockdownd_client is still connected to device (or it thinks it's connected) and has opened session to device, it will first send "goodbye" message to device, and afterwards it releases session and finally releases all resources. in case device already disconnected (either just lockdown service, or device is physically disconnected), this will cause waiting for communication timeout.

if you release client->session_id before calling free, it will not send "googbye", but it might leave opened sockets on device, so i would not recommend that. then again, device will close socket automatically within 60 seconds, so unless you spam it with lockdown connect requests, it might not cause big issues.

AiXanadu, 2023-08-17:
If it's just waiting 60 seconds, it's not a big deal and I'm fine with that.
However, when I use lockdownd to open the "com.apple.mobile.diagnostics_relay" service, and call it in a loop every 5 seconds
```
diagnostics_relay_query_ioregistry_entry(diagnostics_relay, nullptr, "IOPMPowerSource", plist);
lockdownd_get_value(lockdownd, "com.apple.mobile.battery", "BatteryCurrentCapacity", plist);
lockdownd_get_value(lockdownd, "com.apple.mobile.battery", "FullyCharged", plist);
lockdownd_get_value(lockdownd, "com.apple.mobile.battery", "BatteryIsCharging", plist);
```
When obtaining battery information, if the device is suddenly physically disconnected at this time. I cannot release the lockdown.
And at this time the CPU usage increases, on my computer (Intel(R) Core(TM) i9-10900K CPU @ 3.70GHz
) can even reach an occupancy rate of 5%.
If I connect again and disconnect again, it is 5% again. After a few times of unplugging and plugging, the CPU usage is already full.

As for the device's socket, the device is physically disconnected at this point, so there's nothing I can do about it but let it destroy itself.

I read the code, but I don't understand it very well. If possible, please tell me, how can I make sure that the call to lockdownd_client_free or other ***_client_free will not block and occupy the CPU? Or just not using CPU is also possible.

AiXanadu, 2023-08-17:
Did the driver (Windows) actually disconnect the socket because the device disconnected?
Then the lockdown actually calls idevice_connection_send to send the end event to the device, so the block is here?
Since I'm compiling with MinGW32, and calling it with MSVC, I can't debug into the internals of the libimd library. Everything can only be guessed.

mexmer, 2023-08-17:
driver is so low, it doesn't even know there are sockets, sockets are managed by apple mobile device service (which has similar function like usbmuxd on osx/linux), as if they properly disconnect and notify all sockets upon device disconnection, i have no idea, but i doubt it, considering even itunes, when you are doing itunes backup or sync, doesn't interrupt it immediately, when you disconnect devices, but only after some delay.

AiXanadu, 2023-08-18:
Do you mean that the AppleMobileDeviceService also blocks waiting for a response when the device is disconnected?
But why does the CPU usage increase? I can wait, but if this will cause my CPU usage to increase, I have to create a new process to do this, and when the device is disconnected, I force the process to end so that I don't affect other people's computers.
Is there any way to make it disconnect without hogging the CPU? Don't worry about the things on the device side, because it has been physically disconnected at this time, and no more operations can be done.

mexmer, 2023-08-18:
socket waiting doesn't increase cpu usage, you most likely have error in your code somewhere else, like nonblocking async operation, that repeats after error (which you get in noblocking state immediately) or other error.

as for what AMD service does, i have no idea, it's apple software, not mine, it's just mine observation, that it will sometimes fail immediately, but in most cases tcp socket opened to device doesn't close immediately after device is diconnected but linger for seconds, or even half minute, before it closes.

if you want to speedup operations (what opens on device connect/disconnect), you can try to register callback with idevice_event_subscribe, it should be called when device is connected/disconnected physically, or rather when usbmuxd (or in case of windows AMD services) notices it's connection/disconnection.

AiXanadu, 2023-08-18:
No no no, I'm not doing anything in other threads.
Because I started a single-threaded program, it only does one thing, which is to read battery-related information.
Then when the read fails, call lockdownd_client_free.
But it gets stuck here and the CPU usage goes up.

mexmer, 2023-08-18:
post code, because if you doing only simple operation, like querying battery state (either from lockdown values, or from mobile gestalt), i see no reason to be doing anything weird, unless you spam device with requests. in our app we query battery, and other status value every 10 sec, which is more than enough.

AiXanadu, 2023-08-19:
```

// 电池信息
void XConsoleApplication::deviceBatteryLoop() noexcept
{
	auto		handle_device = static_cast<idevice_t>(nullptr);
	auto		handle_lockdownd = static_cast<lockdownd_client_t>(nullptr);
	auto		handle_service_desc = static_cast<lockdownd_service_descriptor_t>(nullptr);
	auto		handle_diagnostics_relay = static_cast<diagnostics_relay_client_t>(nullptr);

	auto		error_device = IDEVICE_E_UNKNOWN_ERROR;
	auto		error_lockdownd = LOCKDOWN_E_UNKNOWN_ERROR;
	auto		error_diagnostics_relay = DIAGNOSTICS_RELAY_E_UNKNOWN_ERROR;
	auto		device_udid = "00008101-001C11943698001E";

	std::printf(u8"LINE:%d : ----------\n", __LINE__);
	idevice_set_debug_level(1);

	error_device = idevice_new(&handle_device, device_udid);
	if (IDEVICE_E_SUCCESS == error_device && handle_device)
	{
		error_lockdownd = lockdownd_client_new_with_handshake(handle_device, &handle_lockdownd, "BatteryLoop");
		if (LOCKDOWN_E_SUCCESS == error_lockdownd && handle_lockdownd)
		{
			if(lockdownd_start_service(handle_lockdownd, "com.apple.mobile.diagnostics_relay", &handle_service_desc) == LOCKDOWN_E_SUCCESS)
			{
				error_diagnostics_relay = diagnostics_relay_client_new(handle_device, handle_service_desc, &handle_diagnostics_relay);
				lockdownd_service_descriptor_free(handle_service_desc);
				if(DIAGNOSTICS_RELAY_E_SUCCESS == error_diagnostics_relay)
				{
					// ---------- ---------- ---------- ---------- ----------
					while(true)
					{
						plist_t 	vNodeIOR = nullptr;
						error_diagnostics_relay = diagnostics_relay_query_ioregistry_entry(handle_diagnostics_relay, nullptr, "IOPMPowerSource", &vNodeIOR);
						if(DIAGNOSTICS_RELAY_E_SUCCESS == error_diagnostics_relay && vNodeIOR)
						{
							auto		vIORegistry = plist_dict_get_item(vNodeIOR, "IORegistry");
							plist_free(vNodeIOR);
							if(vIORegistry)
							{
								plist_t 	vNodeL1 = nullptr;
								plist_t 	vNodeL2 = nullptr;
								plist_t 	vNodeL3 = nullptr;
								auto		vErrorL1 = lockdownd_get_value(handle_lockdownd, "com.apple.mobile.battery", "BatteryCurrentCapacity", &vNodeL1);
								auto		vErrorL2 = lockdownd_get_value(handle_lockdownd, "com.apple.mobile.battery", "FullyCharged", &vNodeL2);
								auto		vErrorL3 = lockdownd_get_value(handle_lockdownd, "com.apple.mobile.battery", "BatteryIsCharging", &vNodeL3);

								plist_free(vNodeL1);
								plist_free(vNodeL2);
								plist_free(vNodeL3);
								if(LOCKDOWN_E_SUCCESS != vErrorL1 || LOCKDOWN_E_SUCCESS != vErrorL2 || LOCKDOWN_E_SUCCESS != vErrorL3)
								{
									std::printf(u8"LINE:%d : ----------\n", __LINE__);
									break;
								}
							}
							else
							{
								std::printf(u8"LINE:%d : ----------\n", __LINE__);
								break;
							}
						}
						else
						{
							std::printf(u8"LINE:%d : ----------\n", __LINE__);
							break;
						}

						// Sleep(5000) or sleep(5)
						x_posix_msleep(5 * 1000);
					}
					// ---------- ---------- ---------- ---------- ----------

					std::printf(u8"LINE:%d : ----------\n", __LINE__);
					diagnostics_relay_client_free(handle_diagnostics_relay);
					std::printf(u8"LINE:%d : ----------\n", __LINE__);
				}
			}
			std::printf(u8"LINE:%d : ----------\n", __LINE__);
			lockdownd_client_free(handle_lockdownd);
			std::printf(u8"LINE:%d : ----------\n", __LINE__);
		}
		std::printf(u8"LINE:%d : ----------\n", __LINE__);
		idevice_free(handle_device);
		std::printf(u8"LINE:%d : ----------\n", __LINE__);
	}
	// system("PAUSE");
	x_posix_system("PAUSE");
}

```

The above is my execution code, when I unplug the device, it will be blocked in lockdownd_client_free.

[image]
[image]
[image]

mexmer, 2023-08-19:
hmm i don't see any problem with your code, but also don't observe same problem, as you describe, i have compiled it in msys2.

just one suggestion to your code (it's what we do in our code), free lockdown client after you start service, you since don't need it anymore.

it's general rule ... unless you need to start multiple services, or read multiple values from lockdown, you just 

connect lockdown 
start service or read/write lockdown value
disconnect lockdown

if you have started service, then service client persist after lockdown is disconnected, so there is no reason to keep lockdown connected.

AiXanadu, 2023-08-19:
As you said, I do this elsewhere, I release the lockdownd_client_free , but since I need to read some status here, I don't.

I compiled it with MinGW, so I don't know what is the difference.
Is there a way I can make it disconnect immediately?

AiXanadu, 2023-08-19:
Is there only one port for lockdownd? Is it possible to connect and disconnect infinitely?

mexmer, 2023-08-19:
lockdown has only one port, as for connection limit, i have no idea,
our software simply emulates behavior of itunes. i did some analysis of itunes communication trough usb, and also some reverse engineering on calls to AMD service, and it never does multiple lockdown connection, so we constructed our software same way, that there is always only one connection with lockdown, that is always used only for necessary time, and then disposed.

AiXanadu, 2023-08-21:
Can it be understood that it is not recommended to run multiple Lockdownd concurrently?
Instead, only make Lockdownd connections when needed?
But if I need to get the battery charge status, then I need to connect to Lockdownd frequently.
```
lockdownd_get_value(handle_lockdownd, "com.apple.mobile.battery", "BatteryIsCharging", &plist);
```

mexmer, 2023-08-21:
it's not that battery status is charging ultrafast, if you poll it's status every 5 second, it's more than enough. you don't need to update status in realtime and poll 10 the value times per second.

maybe you can subscribe for notification for battery change status, no idea (i had not played much with notification proxy, except backup and sync notifications).

i would recommend getting any usb monitoring device, and check communication between apple computer, or windows computer that is running itunes, to check, how often they query device,

AiXanadu, 2023-08-21:
Checking others is not useful, because my requirement here is to update in near real time (every 5 seconds).
It's just that I'm not sure whether to always turn on Lockdownd to read the value every 5 seconds, or every 5 seconds (turn it on for a read and turn it off).

AiXanadu, 2023-08-21:
Of course, I read more detailed battery data from the diagnostic relay, but the state of charge is read by Lockdownd.

The solution you mentioned has been experimented before. When the device is connected for a long enough time (more than half an hour), it may cause the device to fail to connect. Probably because the device socket resource has been exhausted.

mexmer, 2023-08-21:
i noticed in your loop you always call idevice_new, that should not be necessary, since device_handle should be persistent until device is unplugged.

in our application we have callback on notification, when we get device arrival notification, we create proxy device object, that acquires devicehandle with following function (since we only want usb connected devices, not wireless).
idevice_new_with_options(&device_handle, udid, IDEVICE_LOOKUP_USBMUX)

handle is then stored and used all time, until program ends, or device is unplugged.
in rare cases, when we get usbmux error, we also release device handle, and acquire new, but it's usually not necessary.

never had issues running our software even days, because when you root device, full file system extraction takes literally hours, on 1TB that is almost full, it can even take full day. mind iPhone supports only USB 2.0 transfer rates (which is theoretically 480Mbps, but practically it's like 350Mbps), and even most ipads (except M1 based) support only USB 2.0 transfer rates. only USB-C equiped iPad pro has higher speeds.

our app checks device states for various values (like current time, battery status, and such) every 10 seconds.

AiXanadu, 2023-08-21:
Please ignore idevice_new_with_options, because I'm using a separate process to get it, so I need to connect it by UDID (there may be multiple devices connected, I need to know which one the process needs to handle).

You mean, in your application, every 10 seconds to get the value of various possible changes of the device, no problem is it?
The process is: connect-acquire-disconnect. every 10 seconds. connect-acquire-disconnect. Is that so?

I need to connect WIFI devices because the app is designed for devices that need to handle WIFI connections.

I've tested before on MacOS, when I take the value every 5 seconds (connect-acquire-disconnect), after half an hour (or more) I won't be able to connect to the device again until I restart the app. I'm not sure where the problem is.

mexmer, 2023-08-21:
yes i mean, every 10 second we do what you do in loop. we check values in diagnostic relay, and mobile_gestalt service too, and there seems to be no problem with that.

mind every call to idevice_new (or idevice_new_with_options), creates new local connection from your app to usbmuxd (or on windows apple mobile device service), so it might be also exhaustion of local sockets and not device sockets. that's why i suggest "caching" device handles, as better approach, and react to device connect/removal notification accordingly.

AiXanadu, 2023-08-21:
Do you mean that (idevice_new_with_options >> usbmuxd_get_device >> usbmuxd_get_device_list) will cause me to exhaust the resources the system allows me to use?

mexmer, 2023-08-21:
if you do it too often, it might be the possibility.

usbmuxd_get_device_list opens socket to usbmuxd, then closes it
usbmuxd_get_device opens socket to usbmuxd, which you actually use to communicate with device (usbmuxd acts as proxy).

sockets are not close immediately, they always are for short time in wait_for_close state. 

if you cache idevice_handle, socket is never closed, until you call idevice_free, so you don't unnecessary open new sockets between your application and ubsmuxd.

it's one of the things i remind our programmers too, if you use same resource over and over, try to keep it persitent if it's possible, instead of periodically allocating it and dispozing it, when you communicate with device or service, you should optimize your approach, keep opening and closing connection at reasonable rate.

for example with android phones, we open only one shell connection, and use it for all commands, it's better resource wise, than opening terminal every time we need to run command ... of course, there are cases, when this approach is not suitable, but in most cases it's better this way, same goes for sync link (which is used for file transfer).

AiXanadu, 2023-08-21:
Thanks for your guidance.
Although the situation that caused me to not be able to connect to the device can no longer be reproduced, I think this may be the reason.

It's a pity that Lockdownd_client_free is only found on my computer, which leads to an increase in CPU usage. Maybe other people's computers also have it, and this problem cannot be solved for the time being.