libimobiledevice/libimobiledevice #1243: A question about multithreading

xanadu-lee, 2021-11-22:
I would like to ask a question. The applications you currently provide, such as ideviceinfo, are based on single thread.
When I use multithreaded calls, for example, I enable a lockdownd and call lockdown_get_value multithreaded, will there be a problem?
At present, I always receive the error prompt of LOCKDOWND_E_MUX_ERROR.
Moreover, the lockdown_client_free sometimes gets stuck, so I encountered difficulties.
I can only choose not to release and reopen a lockdown.
Can you tell me where to improve?
So that I can use your library better?

nikias, 2021-11-22:
If you want to use multiple threads you need to make separate connections for them. If for example you want to access lockdownd from two threads you need to basically call `lockdownd_client_new` or `lockdownd_client_new_with_handshake` in both of them, so each has its own connection.

xanadu-lee, 2021-11-23:
https://github.com/libimobiledevice/libimobiledevice/issues/1241
How to solve this problem?
Could it be an SSL problem? When I try to close lockdown, how can I quickly exit and free memory?
Generally speaking, when a socket is closed, all sending and receiving should fail and return.
But I didn't get the result as I thought.
Maybe I misunderstood something.

xanadu-lee, 2021-11-23:
In addition, if I open and close lockdown frequently (once every 3 seconds), will there be any problems?

mexmer, 2021-11-23:
@xanadu-lee depending on device, it might refuse connection if you open session to it too offten. you can keep lockdown connection (session) open for extended period of time, and start multiple services or do multiple queries to different values. one lockdown session should not exceed one minute tho' from my experience.

also TCP connection to device is always closed from device side, regardless you send bye message, or TCP reset, it will always take some time (roughly 30-60 sec) before device actually closes connection and releases handle to connection, which means it service inside device self limits how offten you can connect to it.

in our software when we open connection to lockdown, we leave it open for short period of time, in case software needs to do another request to lockdown, to avoid burst of lockdown connections ... there are few exceptions from this rule, but in general we try to keep lockdown connection up to 30 seconds (or 5 seconds idle), before closing it. on older devices and ios version, it's quite easy to exhaust device sockets, and it will simply stop responding, unless you replug it, or unless you wait some time and it starts closing all requests on it's own and free up descriptors.

as for multithreading, i don't recommend accessing one device from multiple threads, while some services can be multiplexed (for example sync and afc), some cannot (afc and housearrest), and it can lead to complete halt of communication with device. you can have multithreaded software, but you should avoid paralel requests to same device. there is no issue with access multiple devices at same time trough usbmux, each from own thread tho'.

xanadu-lee, 2021-11-23:
@mexmer Thank you very much for your explanation.
I tried concurrent lockdown once and started the services respectively, but I didn't find any problems for the time being.
According to your opinion, I should open the service when I need it and close it when I don't need it, right?
For example, if I want to view a directory list, I need to lock a lockdown, start AFC, and then close it?
Restart when I want to query the specific information of the file? I think it's a little too troublesome.
Can you tell me how many lockdownds can be started in one minute?
If I don't close lockdown for a long time, will there be any problems. For example, blocking.

mexmer, 2021-11-23:
AFC is exception, after device is connected, you start it once, then keep it open ... but you can disconnect lockdown, after you get hold of AFC connection. 
other services, like diagnostics, notification proxy, sync, should be closed, when you don't need them.

xanadu-lee, 2021-11-23:
@mexmer Thank you for your reply.
I know how to use it now. your explanation was very detailed.

Now I have another question.
I start the ’com.apple.springboardservices‘ service every 20 seconds, almost every 1-2 hours or so, I will receive a bus error:10 on the mac. when starting the WiFi device and calling lockdownd_start_service.
At first, it collapsed in libimobiledevice-glue:socket_connect_addr, i thought it was a network device problem, so I modified it to use USB connection, now it crashes again in libusbmuxd:send_packet:550(may not be accurate).
If there is no solution, I will have to call it in another process, so that my main process will not be affected by the collapse.

xanadu-lee, 2021-11-23:
see https://github.com/libimobiledevice/libimobiledevice-glue/issues/12

mexmer, 2021-11-23:
considering stack trace, that leads to thread management library inside kernel, it might not be related to libimobile
another possibility is, that you overwrite or release pointer you sent inside in another of your threads. i would first do test with single threaded application, if it's stable, then it's time to try it with multiple threads.

while we use springboard service in our app, we query it only few times during whole application lifetime, so i din't encounter this issue.

xanadu-lee, 2021-11-23:
I create lockdown in a separate thread, start the service, and close lockdown after obtaining the image. I have modified most of my code as you described, including images.
Now they turn lockdown on and off based on a single function

xanadu-lee, 2021-11-24:
I looked at the connection logic again. Can I think that when I successfully connect to the service, I can disconnect lockdown and keep only the service port handle?
I see that the service is communicating using a new socket port.

nikias, 2021-11-24:
> I looked at the connection logic again. Can I think that when I successfully connect to the service, I can disconnect lockdown and keep only the service port handle?
> I see that the service is communicating using a new socket port.

Yes. Especially lockdownd doesn't keep the connection open more than 10 seconds anyway, so whatever you want to do, just start a new lockdown connection every time. iTunes and all other Apple software does this too.

xanadu-lee, 2021-11-24:
@nikias 
Thank you for your reply.

I found that on the Mac, after a long time (more than 1 hour), when I connect lockdown, I will return muxerror, reopen the application, and the connection is normal. What's the reason.

Is the socket not fully released? I call lockdown_client_free every time.

closed 2023-05-16 by xanadu-lee