libimobiledevice/libimobiledevice #1244: unable to install from --HEAD

Karengrigoryan060, 2021-11-23:
Unable to install libimobiledevice from head on macOS. 

brew install --HEAD libimobiledevice

getting error:
No package 'libimobiledevice-glue-1.0' found

kozyraP, 2021-11-26:
I am having the same issue. I created turnaround for myself. You can try if it is gonna work for you:

`brew tap kozyrap/libimobiledevice`
`brew install kozyrap/libimobiledevice/libimobiledevice --HEAD`

I created a fork from libimobiledevice main repo and I changed Head to be as last commit that doesn't need libimibiledevice-glue as a dependency. Then I created brew tap for that fork.
[tap](https://github.com/kozyraP/homebrew-libimobiledevice)
[repo url](https://github.com/kozyraP/libimobiledevice)

Another solution that should work is to copy working, already installed libimobiledevice from another Mac, zip it (it should be named for example as "HEAD-16a8b97")  and unzip it into your folder:
`/usr/local/Cellar/libimobiledevice`
and than run command:
'brew link libimobiledevice'

Both solution are a bit messy but nothing else worked for me.

karengrigoryanpicsart, 2021-11-28:
@kozyraP yes, this is working, but the problem is I need to use ios_webkit_debug_proxy, and with this version, the proxy is not connecting. Do you have any idea how I can fix it?

kozyraP, 2021-11-28:
@karengrigoryanpicsart  
Yea, I had same problem with ios-webkit-debug-proxy and second solution resolve problem for me.

mexmer, 2021-11-29:
you can check this thread https://github.com/libimobiledevice/libimobiledevice/issues/1217
i always compile from source, so have no solution, but seems people already did something around.

realMatthewBenedict, 2022-09-17:
Thanks for the link!

I was finally able to install thanks to this specific comment: https://github.com/libimobiledevice/libimobiledevice/issues/1217#issuecomment-1009402345

UInt2048, 2022-09-17:
Thanks for the link!

I was finally able to install thanks to this comment: https://github.com/libimobiledevice/libimobiledevice/issues/1217#issuecomment-1009402345