libimobiledevice/libimobiledevice #1248: Socket API mismatch between libimobiledevice and libusbmuxd

MarSoft, 2021-12-06:
The code in `libimobiledevice` expects that `libusbmuxd` will provide `conn_data` for networked devices in the following format:
- byte 0: number of meaningful bytes in the 200-byte buffer [ref](https://github.com/libimobiledevice/libimobiledevice/blob/master/src/idevice.c#L281)
- byte 1: address family as provided by MacOS (0x02 or 0x1E) [ref](https://github.com/libimobiledevice/libimobiledevice/blob/master/src/idevice.c#L475-L479)
- bytes from 2: 14 or 26 bytes containing address itself.

Now, if we peek into `libusbmuxd` code, it has different logic:
- take what `usbmuxd` gave us in plist data for `NetworkAddress`;
- put it into the buffer, unless it exceeds buffer size. [ref](https://github.com/libimobiledevice/libusbmuxd/blob/master/src/libusbmuxd.c#L288)

And looking ad the data provided by `iTunes` implementation of `usbmuxd`, it looks as follows (for IPv4 addresses):
```
02 00 00 00    // address family, 4 bytes, little endian
0c a8 00 8e    // hex representation of 192.168.0.142 IP address
00 00 00 00 00 00 00 00 ...   // zero-padding to reach 128 bytes in total
```

So we try to interpret first byte of address family (`0x02`) as data size, and second byte of address family (`0x00`) as address family value. This leads to `Unsupported address family 0x00` error message, and makes it impossible to connect iDevices over network.

***Possible solution:***
1. Change `libusbmuxd` to include data size as a first byte;
2. Change `libimobiledevice` to look for IP address data at the right offset (5, not 2: skip 1 byte for length and 4 bytes for family).

***Possible workaround*** is proposed in https://github.com/libimobiledevice/libusbmuxd/issues/95#issuecomment-714297655 - we can patch `libusbmuxd` to report Networked devices as if they are USB ones, forcing `libimobiledevice` to proxy connections to them via `usbmuxd` (as it was before 2c83cae88c50217f0fbbff5c3d2eab5f252f164e). I don't like this particular workaround implementation tough because it makes impossible to distinguish between networked and USB devices for client code.

xanadu-lee, 2022-01-05:
https://github.com/libimobiledevice/libusbmuxd/pull/121

beni20, 2023-05-11:
> [libimobiledevice/libusbmuxd#121](https://github.com/libimobiledevice/libusbmuxd/pull/121)
worked for me. Without it restore mode device was not detected

xanadu-lee, 2023-05-11:
If there is a problem, you can force a USB connection, and the WIFI forwarding is handled by the driver.

nikias, 2023-06-27:
I pushed commit a172604e5afaded8f0db1eb765be443984752400 which would just take the buffer that we get from libusbmuxd. It's the platform-dependent address that should just allow to connect.