libimobiledevice/libimobiledevice #1252: Unreliable installation guide

fohfuu, 2021-12-08:
As was the cause of #1240, #1112, #1111, #1032, #1028, #1013, #1001 and #1115, the installation guide in the README is directing users to install all dependencies using apt-get, which intermittently causes build failures and errors.

This is not a user error; it's the result of users doing what they were told.

Editing the installation guide to remove packages that suffer this problem from the apt step and directing users where to install from upstream would permanently solve the issue.

fohfuu, 2022-01-17:
Since I've received no reply either way to this, I want to offer a different solution.

How about adding a line to the README that states the instructions only work up to iOS 13, and dependencies should not be installed using apt for use with iOS 14/15.

Nothing more, no need do any legwork. Just a simple sentence until the next release.

SirZayers, 2022-01-29:
I second this.
First of all the website, although nicely designed doesn't really give any explanation on how to use the software, not does it even list all the major distros that are being maintained, I for one was surprised to find out that I actually had it installed on my Arch system.
And secondly, please provide some form of instruction on how to build the thing yourself on different distros, at least until the maintainers have pushed their respective updates, and a message to explain new users that they have to build from source, otherwise this issue tracker will be flooded with the same question over and over again,
I would try to help, but as I can't run the thing I can't figure out how it works to write anything about it.

dickorydock, 2022-03-07:
I'm new to idevicebackup2 and found this issue while troubleshooting. Please do this! I am actually still having issues getting it to work even after trying what I found in this thread: https://askubuntu.com/questions/1297635/ubuntu-libimobiledevice-ios-14-focal-20-04. If you could possibly add instructions here (or link) for how to install from github I would really appreciate it.