libimobiledevice/libimobiledevice #1257: Failed to backup with iOS 15.2

baochenhu, 2021-12-17:
./idevicebackup2 backup  ./myIOS

Backup directory is "./myIOS"
Started "com.apple.mobilebackup2" service on port 56216.
Negotiated Protocol Version 2.1
Starting backup...
Backup will be encrypted.
Requesting backup from device...
Full backup mode.
ErrorCode 3: Failed to create snapshot: File exists (17) at path "/private/var" (MBErrorDomain/3)
Received 0 files from device.
Backup Failed (Error Code 3).

pixielark, 2021-12-18:
I run into the same issue probably due to I ctl+c the previous backup
solution is to reboot your iDevice and I was able to proceed

baochenhu, 2021-12-18:
yeah, after rebooting, it seems now I see a different error.

./idevicebackup2 backup  ./myIOS

Backup directory is "./myIOS"
Started "com.apple.mobilebackup2" service on port 49230.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Backup will be encrypted.
Requesting backup from device...
Incremental backup mode.
ERROR: Could not receive from mobilebackup2 (-4)
Received 0 files from device.
Backup Aborted.

Have you seen this before?

Arr0way, 2022-06-28:
Yes, identical issue here also. 

@baochenhu were you able to resolve?

mexmer, 2022-06-28:
```
ErrorCode 3: Failed to create snapshot: File exists (17) at path "/private/var" (MBErrorDomain/3)
```
this error is from device itself, and is caused by remnants of unfinished backup/restore ... this usually can be solved by rebooting phone.

```
ERROR: Could not receive from mobilebackup2 (-4)
```
as for this, looks like device is either not communicating, or communication did time out.
this can be caused by different reasons, for example insufficient storage or memory to create backup in device, or there is just too many files to process, and device stopped responding after a while, also it can happen if backup process inside device crashes

check device syslog, if you see there some backup service related error.

mexmer, 2022-06-28:
yes, you do have access to syslog, if you can run idevicebackup2, you can also run idevicesyslog

in one terminal session run 
```
idevicesyslog
```
in other terminal session, run backup