libimobiledevice/libimobiledevice #1259: usbmuxd_recv_timeout returned -108 (connection reset)

P5-2005, 2021-12-26:
Hello,

trying to ideviceinfo on watchos 6.3, i got this error : 

08:55:50 D:\a\1\s\libimobiledevice\src\idevice.c:587 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -108 (connection reset)
08:55:50 D:\a\1\s\libimobiledevice\src\service.c:167 service_receive_with_timeout(): could not read data
08:55:50 D:\a\1\s\libimobiledevice\src\property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!

here debug of idevicepair :

======================

`C:\Users\BH11040\Downloads\utils>idevicepair pair -d
08:59:11 D:\a\1\s\libimobiledevice\src\lockdown.c:650 lockdownd_client_new(): device udid: 2b3e0b6ec555e5a575802ea7404f81150e6edac5
08:59:11 D:\a\1\s\libimobiledevice\src\lockdown.c:379 lockdownd_query_type(): called
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:132 internal_plist_send(): sending 284 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\service.c:144 service_send(): sending 4 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\idevice.c:545 idevice_connection_send(): internal_connection_send 4, sent 4
08:59:11 D:\a\1\s\libimobiledevice\src\service.c:144 service_send(): sending 284 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\idevice.c:545 idevice_connection_send(): internal_connection_send 284, sent 284
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:137 internal_plist_send(): sent 284 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicepair</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 297 bytes following
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 297 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
08:59:11 D:\a\1\s\libimobiledevice\src\lockdown.c:395 lockdownd_query_type(): success with type com.apple.mobile.lockdown
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:132 internal_plist_send(): sending 333 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\service.c:144 service_send(): sending 4 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\idevice.c:545 idevice_connection_send(): internal_connection_send 4, sent 4
08:59:11 D:\a\1\s\libimobiledevice\src\service.c:144 service_send(): sending 333 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\idevice.c:545 idevice_connection_send(): internal_connection_send 333, sent 333
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:137 internal_plist_send(): sent 333 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:138 internal_plist_send(): printing 333 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicepair</string>
        <key>Key</key>
        <string>DevicePublicKey</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 914 bytes following
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 914 bytes
08:59:11 D:\a\1\s\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 914 bytes plist:
[... 173 lines cut ...]
        <key>PairingOptions</key>
        <dict>
                <key>ExtendedPairingErrors</key>
                <true/>
        </dict>
</dict>
</plist>
08:59:12 D:\a\1\s\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
08:59:12 D:\a\1\s\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 231 bytes following
08:59:12 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 231 bytes
08:59:12 D:\a\1\s\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 231 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>Pair</string>
</dict>
</plist>
08:59:12 D:\a\1\s\libimobiledevice\src\lockdown.c:956 lockdownd_do_pair(): Pair success
08:59:12 D:\a\1\s\libimobiledevice\src\lockdown.c:958 lockdownd_do_pair(): internal pairing mode
08:59:12 D:\a\1\s\libimobiledevice\src\lockdown.c:973 lockdownd_do_pair(): Saving WiFiAddress from device in pair record
SUCCESS: Paired with device 2b3e0b6ec555e5a57580dddddddde6edac5`

===========================

P5-2005, 2021-12-26:
> test

nikias, 2021-12-26:
Just to be clear, you are trying to connect via USB? Looks like it fails when establishing the device pairing; this might be something the watch doesn't support. Usually the watch has the be paired to an iPhone and then you connect to it through companion proxy.

P5-2005, 2021-12-27:
Yes, im using ibus adapter, last week i tried with watchos 2.1 and iwatch s0(1st gen), even paired with my phone i can ideviceifo no issue
when i tried with this s2 and watchos 6.3 i got this, even paired with my iphone xs ios 15, i tried unpaired statw, same error
i dont know if possible on new watchos or just useless feature 😅

mexmer, 2021-12-29:
> 

do you use correct adapter? there is 3 ibus adapters
ibus S1 - works with watch S0/S1
ibus S2 - works with watch S2/S3
ibus S4 - works with watch S4/S5/SE (doesn't work properly with S6)

P5-2005, 2021-12-29:
> > 
> 
> do you use correct adapter? there is 3 ibus adapters ibus S1 - works with watch S0/S1 ibus S2 - works with watch S2/S3 ibus S4 - works with watch S4/S5/SE (doesn't work properly with S6)

yes, im using magicawrt, the special slot of s2 42mm, used same one for s0 on other slot all good with ideviceinfo

nikias, 2021-12-29:
So generally communication works, even the pairing seems to succeed, right?
I think the problem might be the ValidatePair request. Lockdownd from a certain iOS version on, on iPhones, it doesn't like that and also just drops the connection, so maybe it's that? Actually the code here:
https://github.com/libimobiledevice/libimobiledevice/blob/master/src/lockdown.c#L741
should not call it for watch devices...

P5-2005, 2021-12-29:
> So generally communication works, even the pairing seems to succeed, right? I think the problem might be the ValidatePair request. Lockdownd from a certain iOS version on, on iPhones, it doesn't [...]

yes, pair always got success message with udid of device,
i should build it now from master to see if will work?

nikias, 2021-12-29:
Yes, please compile latest code. Maybe it just works then.

P5-2005, 2021-12-29:
yes, worked fine, thank you so much

closed 2021-12-29 by P5-2005

nikias, 2021-12-29:
Great! Glad it worked.