libimobiledevice/libimobiledevice #1261: iOS 15 devices: idevicescreenshot error. Could not connect to screenshotr!

karthikk007, 2021-12-30:
Context: I raised the same issue on iOS 14 devices long back. https://github.com/libimobiledevice/libimobiledevice/issues/1123

iOS 15 devices: idevicescreenshot error. Could not connect to screenshotr!

I am facing this issue on iOS 15 devices.

I am using mac machine. 
Could you please tag and release the latest version?

Tried the following

**install latest**: not working.
===================

brew uninstall --ignore-dependencies libimobiledevice
brew uninstall --ignore-dependencies ideviceinstaller
brew uninstall --ignore-dependencies usbmuxd
brew uninstall --ignore-dependencies libplist
sudo rm /var/db/lockdown/*
brew install --HEAD libplist
brew install --HEAD usbmuxd
brew unlink usbmuxd
brew link usbmuxd
brew unlink libusbmuxd
brew link libusbmuxd
brew install --HEAD libimobiledevice
brew install --HEAD ideviceinstaller

**Install from git: Facing compiler issue**

git clone https://github.com/libimobiledevice/libimobiledevice.git
cd libimobiledevice

export openssl_LIBS="-L/usr/local/opt/openssl/lib"
export openssl_CFLAGS="-I/usr/local/opt/openssl/include/"

./autogen.sh
make
sudo make install

karthikk007, 2021-12-30:
@nikias @FunkyM could you please help on this?

mexmer, 2021-12-30:
to compile libimobiledevice you need first compile and install libimobiledevice-glue, libplist and libusbmuxd

nikias, 2021-12-30:
On macOS, you can easily build everything with my script:
```bash
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -Ls git.io/JEhrV 
bash ./limd-build-macos.sh
```

karthikk007, 2021-12-31:
I am getting this error

```
[*] Installing...
#### Building libimobiledevice ####
[*] Configuring...
ERROR: configure failed for libimobiledevice, check /Users/kk007/work/libimobile/limd-build/configure.log for details.
```

```
checking for struct dirent.d_type... yes
checking for openssl... no
configure: error: OpenSSL support explicitly requested but OpenSSL could not be found
```

mexmer, 2022-01-03:
you need to install openssl-dev package, it's one of prerequsities

bentumbler, 2022-01-03:
@nikias Your macOS script seems to use both architectures (arm64 and x86_64) in the cflags. Is this building a universal library? Are any further steps needed to create a universal lib?

nikias, 2022-01-03:
It creates fat binaries for both architectures.

closed 2022-01-05 by karthikk007

nikias, 2022-01-05:
Yes that is actually not what should happen, it should use macOS' SSL libs instead, but it fails if there's already openssl installed through homebrew. I am trying to improve this.

D2y3, 2024-12-12:
Did the problem solve？
please tell me how to deal with the problem。