libimobiledevice/libimobiledevice #1265: Library not loaded: /usr/local/lib/libimobiledevice-1.0.6.dylib

kuuu1234, 2022-01-14:
I've been following the macOS instructions and I don't know if I'm doing it wrong or something. https://jkcoxson.github.io/DiOS-Instructions/
[image]

I also tried this method and nothing happens either, another error, can someone help me out? Running macOS 12.1
[image]

I'm basically stuck on on step 4-5

nikias, 2022-01-19:
Don't follow some random instructions. For macOS I always suggest to use my build script:
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L is.gd/kdT2Nl
bash ./limd-build-macos.sh
```

kuuu1234, 2022-01-21:
> Don't follow some random instructions. For macOS I always suggest to use my build script:
> 
> ```
> mkdir -p limd-build
> cd limd-build
> curl -o ./limd-build-macos.sh -L is.gd/kdT2Nl
> bash ./limd-build-macos.sh
> ```
> 
> This installs everything with the most recent code in /usr/local by default.

So then what are my next steps? Or if I were to restart what do I do from step one to finish?

kuuu1234, 2022-01-21:
> Don't follow some random instructions. For macOS I always suggest to use my build script:
> 
> ```
> mkdir -p limd-build
> cd limd-build
> curl -o ./limd-build-macos.sh -L is.gd/kdT2Nl
> bash ./limd-build-macos.sh
> ```
> 
> This installs everything with the most recent code in /usr/local by default.
[image]

Your script did not work. I might be missing files or something. 

The instructions that I saw online was specific and also for AltStore and to download Jit for dolphin or another emulator that required jit I believe.

If you have a FULL set your instructions with links that need any downloading or even a video to download JIT with your script then I would greatly appreciate it.

kuuu1234, 2022-01-22:
@nikias

nikias, 2022-01-23:
You have a typo in the curl command. Just copy and paste it.

kuuu1234, 2022-01-23:
just did, what are the next steps to do? This is for JIT right?

kuuu1234, 2022-01-23:
I'm not sure what exactly I am installing, is this for being able to use JIT on iOS? I got a few warnings that a version of M4 was found and doesn't support -g? It still went through saying I have to do other things. Is there a link to find the instructions on what to do next?
[image]
[image]
Ill wait for your response because I'm not sure what to do next

nikias, 2022-01-24:
What exactly do you mean with "JIT on iOS"? You are building the host conputer libraries and tools, you won't change anything on the device. I am not sure what you are planning to do exactly, since you never said that.
As for the ifuse question, if unsure say N.

kuuu1234, 2022-01-24:
I was trying to play a GameCube or wii game on dolphin emulator or gc emulator, it required JIT and that link I put in the first thread has the instructions on how to do it, I was asking help on why that's happening and how to fix it. By any chance do you have a of a work around for the first screenshots? @nikias

kuuu1234, 2022-01-24:
It's something needed in order to play on the phone, there's an AltStore beta that comes with it pre installed but I don't think it's stable yet

nikias, 2022-01-24:
The first screenshot only shows warnings, no errors, and the second shows that it actually continued to build. I don't see anything wrong. You should now be able to run `idevicedebug`. Also check if `ideviceinfo` works now.

kuuu1234, 2022-01-24:
So this will help me with the question I had earlier?? To use jit. I wasn't able to use idevice_id in the first place which is why i came here so now you suggested another script and this should work? Before I select Y how do I reduce it in recovery mode? What is the command? And it says the kernel extension cache has to be regenerated?

kuuu1234, 2022-01-30:
@nikias

nikias, 2022-01-31:
@kuuu1234 that question you have to answer is about macFUSE. That's a virtual file system driver and only required if you want to use the `ifuse` tool (https://github.com/libimobiledevice/ifuse). It is not required so you can safely say no here and it will just not build ifuse but the rest.
I am not sure about if the JIT stuff will work; in any case after running my script you can just run all the `idevice*` commands without `./`, so just type `idevice_id` or `idevicedebug` or whatever you need to do it says in the linked tutorial.

kuuu1234, 2022-01-31:
I'm trying to figure out why I get that error though, I was asking for help on how to get a solution and if anyone knows that can help

nikias, 2022-01-31:
You need to be more specific. In case you are talking about the "Library not loaded" error, the first comment on this ticket explains it; it tries to find the library in /usr/local/lib while you probably have them extracted somewhere in Downloads. But as the author of the first comment here notes, it won't find libusbmuxd-2.0.6.dylib then. Which is obvious since the tutorial didn't tell you you have to download that too.
That's why I said, run my script, when asked about ifuse, say N, and you will have all conmands ready to use.

kuuu1234, 2022-02-01:
So once I do your script I should be able to use the first script??

kuuu1234, 2022-02-07:
@nikias

nikias, 2022-02-08:
You don't need to @ me, I get an email for every comment made here. After my script you can do what's in their script, just remove the ./ in front of the commands since you now have it installed in a global location.

kuuu1234, 2022-02-09:
Hey thank you for helping me out. Im on the last step of the link I posted, it saying bundle id not found, do you know if its something I did incorrectly? I tried with ./ and / but seems to not work. I have a screenshot, could you maybe review the steps and see if there was anything I did wrong? Thanks again!
[image]

nikias, 2022-02-09:
That doesn't look like a valid bundle identifier to me. Try `ideviceinstaller -l` to see what apps you have installed.

mexmer, 2022-02-09:
@kuuu1234 it's small letter L, not number 1
use copey & paste next time

kuuu1234, 2022-02-09:
Here's what I am seeing, anything unusual?
[image]
[image]

nikias, 2022-02-09:
Come on dude, would it be too much to expect you to think a little? The output is completey normal, you need to find the package ID of the dolphin thingy. It is in the screenshot (pro tip: copy the text of the terminal output instead of a screenshot), it's the one that starts with `me.oatmealdome.` - copy the full app id and paste it to the comman you want to run.

Also please understand this is not a DiOS issue tracker, so with issues you should report them to the original project. Thanks for your understanding.

closed 2022-02-09 by nikias

kuuu1234, 2022-02-09:
Okay thank you ill try that :)