libimobiledevice/libimobiledevice #1268: How to get syscfg, such as NSrN data

bugstacks, 2022-01-20:
I want to get the iPhone's syscfg data, such as NSrN touch-id serial number, etc.
Please see the syscfg data https://www.theiphonewiki.com/wiki/SysCfg

truonggiang0710, 2022-01-27:
@bugstacks you can use the command "idevicediagnostics mobilegestalt SysCfg" to get the data, then decode the data via base64.

bugstacks, 2022-01-27:
> @bugstacks you can use the command "idevicediagnostics mobilegestalt SysCfg" to get the data, then decode the data via base64.

Thanks for the reply, I got the sysCfg data through your way, but I still can't decode it directly through base64, here are my readout values
https://pastebin.com/PFKcX5rn

mexmer, 2022-01-27:
after you decode base64 you get binary blob, you need to decode fields you require from that blob .... no idea, if there is parser for that tho since it's apple proprietary data.

jdbkevinma, 2022-02-15:
hi all
i try to decode the data but have some special characters "\u0005", anyone can help?

gfCS@\u0005\0\0\0\0\u0002\0\u0002\0\u0002\0\0\0\0\0B\0\0\0#BLMF3X83430CR2JVN54MEFRA?\0\0\0}養?\0\0\0c餟KSW\u0001\0\0\0\0\0\0\0JP\0\0\0\0\0\0caMW8S?\0\0\0\0\0\0\0\0\0\0caMB8S?\u0004\b\0\0\0\0\0\0\0\0\0\0BTNCTABC(\u0001\0\0佝\u0001\0BTNCLACB\0\u0011\0\0仨\u0001\0BTNCLACW?\0\0噱\u0001\0mNrSF17X8FUXKPGD\0\0\0\0#GFC////*/*\0\0\0\0\0\0\0\0\0BTNCttaB\u0011\0\0\0体\u0001\0BTNCEFALd\0\0\0h?\0BTNCpMPL?\0\0|?\0BTNClCSPP\u0002\0\0,?\0BTNCLCLCP\0\0\0茑\u0001\0BTNClaCTP\0\0\0屶\u0001\0BTNClCPD(\0\0\0d?\0BTNCLCrP\0\u0001\0\0d?\0BTNClCtM?\0\0茉\u0001\0pSGC\u0001\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0#doMMTAX2\0\0\0\0\0\0\0\0\0\0\0ngeRJ/A\0\0\0\0\0\0\0\0\0\0\0\0\0GCLC\u0001\0\0\0\0\0\0\0\0\0\0\0\u0001\0\0\0SHLC\u0001\0\0\0\0\0\0\0\0\0\0\0\u0002\0\0\0lCoN\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0qSRP\u0001\0\0\u0004\0\0\0\0\0\0\0\0\0\0\0\0BTNCNStM,\0\0\0霸\u0001\0BTNCCldMZ\0\0\0加\u0001\0LCxR\0\u0002?\0\0\0\0\0\0\0\0\0\0\0.BTNCSMCF\u0011\0\0\0溤\u0001\0BTNCBMCF@\0\0\0\\?\0BTNCSMCB\u0011\0\0\0H
?\0BTNCBMCB0\0\0\0\u0018?\0CCVC\0\0\0\0?7\u0001\0\0M\u0001\0\0CCDC\u0001\0\0\0\u0001?邗BTNCOATL\u0014\0\0\0ㄓ\u0001\0BTNCCSRA,\0\0\0|?\0BTNCCSRG,\0\0\0P?\0toRA\u0001\0\0\0\u0001\0\0\0\0\u0001\0\0\0\0\0toRG\u0001\0\0\0\0\0\0\0\0\0\0\0\u0001\0lCIA\u0003\0\0\0}\u0013\0\0?\0\0\0\0\0\0lCIG\u0002\0\0\0}\u0013\0\0?\0\0\0\0\0\0BTNClCTG<\0\0\0\u0014?\0BTNCSArP\u001c\0\0\0\u0001\0BTNCOPPSP\0\0\0ㄒ\u0001\0BTNCNFSL?\0\0\u0004?\0BTNCSTpS,\0\0\0匦\u0001\0BTNCCTRL\u001c\u0003\0\0纪\u0001\0BTNCpSSL?\0\0\u0018?\0BTNCxRSL?\0\0t?\0BTNCMPSL?\0\0腥\u0001\0BTNCCSSL?\0\0,?\0BTNCTTYGb\0\0\0绕\u0001\0BTNClCxP?\0\08?\0BTNCRACB2\0\0\0\u0004?\0BRTS?&*???絬\0\0\0\0MPTPZ013CaMaQaIhSaWaaGiM\u0004\u0002\u0001\0?\u0001\0?\0\0\0\0aGpS\u0002\u0002\u0001\0\u000f\0\0\0\0\0\0\0\0\0\0\0toRC\u0002\0\0\0\0\0\0\u0001\0\0\0\0\0\0BTNCMCSC(\0\0\0芘\u0001\0BTNCLCFL(?\0?\u0001\0BTNCrPSL?\0\0\u00100\u0001\0

jdbkevinma, 2022-02-15:
language:c#

 var result = DecodeBase64(Encoding.Default,"Z2ZDU0AFAAAAAAIAAgACAAAAAABCAAAAI0JMTUYzWDgzNDMwQ1IySlZONTRNRUZSQaUBAAAAffBCpQEAAABj8FVLU1cBAAAAAAAAAEpQAAAAAAAAY2FNVzhTnCqiRwAAAAAAAAAAAABjYU1COFOcNgQIAAAAAAAAAAAAAEJUTkNUQUJDKAEAANj+AQD/////QlROQ0xBQ0IAEQAA2O0BAP////9CVE5DTEFDV/gHAADg5QEA/////21OclNGMTdYOEZVWEtQR0QAAAAAI0dGQy8vLy8qLyoAAAAAAAAAAABCVE5DdHRhQhEAAADM5QEA/////0JUTkNFRkFMZAAAAGjlAQD/////QlROQ3BNUEzsAQAAfOMBAP////9CVE5DbENTUFACAAAs4QEA/////0JUTkNMQ0xDUAAAANzgAQD/////QlROQ2xhQ1RQAAAAjOABAP////9CVE5DbENQRCgAAABk4AEA/////0JUTkNMQ3JQAAEAAGTfAQD/////QlROQ2xDdE2ICgAA3NQBAP////9wU0dDAQAAAAAAAAAAAAAAAAAAACNkb01NVEFYMgAAAAAAAAAAAAAAbmdlUkovQQAAAAAAAAAAAAAAAABHQ0xDAQAAAAAAAAAAAAAAAQAAAFNITEMBAAAAAAAAAAAAAAACAAAAbENvTgAAAAAAAAAAAAAAAAAAAABxU1JQAQAABAAAAAAAAAAAAAAAAEJUTkNOU3RNLAAAALDUAQD/////QlROQ0NsZE1aAAAAvNMBAP////9MQ3hSAAKXOQAAAAAAAAAAAAAALkJUTkNTTUNGEQAAAJzUAQD/////QlROQ0JNQ0ZAAAAAXNQBAP////9CVE5DU01DQhEAAABI1AEA/////0JUTkNCTUNCMAAAABjUAQD/////Q0NWQwAAAACF////NwEAAE0BAABDQ0RDAQAAAKv+//8B/f//2vX//0JUTkNPQVRMFAAAAKjTAQD/////QlROQ0NTUkEsAAAAfNMBAP////9CVE5DQ1NSRywAAABQ0wEA/////3RvUkEBAAAAAQAAAAABAAAAAP8AdG9SRwEAAAD/AAAAAP8AAAAAAQBsQ0lBAwAAAH0TAADIAAAAAAAAAGxDSUcCAAAAfRMAAMgAAAAAAAAAQlROQ2xDVEc8AAAAFNMBAP////9CVE5DU0FyUBwAAAD40gEA/////0JUTkNPUFBTUAAAAKjSAQD/////QlROQ05GU0ykAQAABNEBAP////9CVE5DU1RwUywAAADY0AEA/////0JUTkNDVFJMHAMAALzNAQD/////QlROQ3BTU0ykAQAAGMwBAP////9CVE5DeFJTTKQBAAB0ygEA/////0JUTkNNUFNMpAEAANDIAQD/////QlROQ0NTU0ykAQAALMcBAP////9CVE5DVFRZR2IAAADIxgEA/////0JUTkNsQ3hQkAAAADjGAQD/////QlROQ1JBQ0IyAAAABMYBAP////9CUlRTggAmKvwQpxr0H711AAAAAE1QVFBaMDEzQ2FNYVFhSWhTYVdhYUdpTQQCAQD///z/AQD9/wAAAABhR3BTAgIBAA8AAAAAAAAAAAAAAHRvUkMCAAAAAAD/AAEAAAAA/wAAQlROQ01DU0MoAAAA3MUBAP////9CVE5DTENGTCiUAAC0MQEA/////0JUTkNyUFNMpAEAABAwAQD/");

 public static string DecodeBase64(Encoding encode, string result)
        {
            string decode = "";
            byte[] bytes = Convert.FromBase64String(result);
            try
            {
                decode = encode.GetString(bytes);
            }
            catch
            {
                decode = result;
            }
            return decode;
        }

nikias, 2022-02-15:
Come on man, it's binary data, did you really expect it to be just like a text file?

jdbkevinma, 2022-02-16:
> binary

Dear Nikias

https://pastebin.com/PFKcX5rn

this is the command  "idevicediagnostics mobilegestalt SysCfg" output result,  for "SysCfg", I want to convert it to a string.

jdbkevinma, 2022-02-16:
@nikias

nikias, 2022-02-16:
@jdbkevinma you can't, that's what I am trying to say. It's not a string. It's encoded data in binary form. You need to understand the values and try to make sense of them. Look at the base64 decoded data with a hex editor/viewer: 
```
00000000  67 66 43 53 40 05 00 00  00 00 02 00 02 00 02 00  |gfCS@...........|
00000010  00 00 00 00 42 00 00 00  23 42 4c 4d 46 33 58 38  |....B...#BLMF3X8|
00000020  33 34 33 30 43 52 32 4a  56 4e 35 34 4d 45 46 52  |3430CR2JVN54MEFR|
00000030  41 a5 01 00 00 00 7d f0  42 a5 01 00 00 00 63 f0  |A.....}.B.....c.|
00000040  55 4b 53 57 01 00 00 00  00 00 00 00 4a 50 00 00  |UKSW........JP..|
00000050  00 00 00 00 63 61 4d 57  38 53 9c 2a a2 47 00 00  |....caMW8S.*.G..|
00000060  00 00 00 00 00 00 00 00  63 61 4d 42 38 53 9c 36  |........caMB8S.6|
00000070  04 08 00 00 00 00 00 00  00 00 00 00 42 54 4e 43  |............BTNC|
00000080  54 41 42 43 28 01 00 00  d8 fe 01 00 ff ff ff ff  |TABC(...........|
00000090  42 54 4e 43 4c 41 43 42  00 11 00 00 d8 ed 01 00  |BTNCLACB........|
000000a0  ff ff ff ff 42 54 4e 43  4c 41 43 57 f8 07 00 00  |....BTNCLACW....|
000000b0  e0 e5 01 00 ff ff ff ff  6d 4e 72 53 46 31 37 58  |........mNrSF17X|
000000c0  38 46 55 58 4b 50 47 44  00 00 00 00 23 47 46 43  |8FUXKPGD....#GFC|
000000d0  2f 2f 2f 2f 2a 2f 2a 00  00 00 00 00 00 00 00 00  |////*/*.........|
000000e0  42 54 4e 43 74 74 61 42  11 00 00 00 cc e5 01 00  |BTNCttaB........|
000000f0  ff ff ff ff 42 54 4e 43  45 46 41 4c 64 00 00 00  |....BTNCEFALd...|
00000100  68 e5 01 00 ff ff ff ff  42 54 4e 43 70 4d 50 4c  |h.......BTNCpMPL|
00000110  ec 01 00 00 7c e3 01 00  ff ff ff ff 42 54 4e 43  |....|.......BTNC|
00000120  6c 43 53 50 50 02 00 00  2c e1 01 00 ff ff ff ff  |lCSPP...,.......|
00000130  42 54 4e 43 4c 43 4c 43  50 00 00 00 dc e0 01 00  |BTNCLCLCP.......|
00000140  ff ff ff ff 42 54 4e 43  6c 61 43 54 50 00 00 00  |....BTNClaCTP...|
00000150  8c e0 01 00 ff ff ff ff  42 54 4e 43 6c 43 50 44  |........BTNClCPD|
00000160  28 00 00 00 64 e0 01 00  ff ff ff ff 42 54 4e 43  |(...d.......BTNC|
00000170  4c 43 72 50 00 01 00 00  64 df 01 00 ff ff ff ff  |LCrP....d.......|
00000180  42 54 4e 43 6c 43 74 4d  88 0a 00 00 dc d4 01 00  |BTNClCtM........|
00000190  ff ff ff ff 70 53 47 43  01 00 00 00 00 00 00 00  |....pSGC........|
000001a0  00 00 00 00 00 00 00 00  23 64 6f 4d 4d 54 41 58  |........#doMMTAX|
000001b0  32 00 00 00 00 00 00 00  00 00 00 00 6e 67 65 52  |2...........ngeR|
000001c0  4a 2f 41 00 00 00 00 00  00 00 00 00 00 00 00 00  |J/A.............|
000001d0  47 43 4c 43 01 00 00 00  00 00 00 00 00 00 00 00  |GCLC............|
000001e0  01 00 00 00 53 48 4c 43  01 00 00 00 00 00 00 00  |....SHLC........|
000001f0  00 00 00 00 02 00 00 00  6c 43 6f 4e 00 00 00 00  |........lCoN....|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 71 53 52 50  |............qSRP|
00000210  01 00 00 04 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000220  42 54 4e 43 4e 53 74 4d  2c 00 00 00 b0 d4 01 00  |BTNCNStM,.......|
00000230  ff ff ff ff 42 54 4e 43  43 6c 64 4d 5a 00 00 00  |....BTNCCldMZ...|
00000240  bc d3 01 00 ff ff ff ff  4c 43 78 52 00 02 97 39  |........LCxR...9|
00000250  00 00 00 00 00 00 00 00  00 00 00 2e 42 54 4e 43  |............BTNC|
00000260  53 4d 43 46 11 00 00 00  9c d4 01 00 ff ff ff ff  |SMCF............|
00000270  42 54 4e 43 42 4d 43 46  40 00 00 00 5c d4 01 00  |BTNCBMCF@...\...|
00000280  ff ff ff ff 42 54 4e 43  53 4d 43 42 11 00 00 00  |....BTNCSMCB....|
00000290  48 d4 01 00 ff ff ff ff  42 54 4e 43 42 4d 43 42  |H.......BTNCBMCB|
000002a0  30 00 00 00 18 d4 01 00  ff ff ff ff 43 43 56 43  |0...........CCVC|
000002b0  00 00 00 00 85 ff ff ff  37 01 00 00 4d 01 00 00  |........7...M...|
000002c0  43 43 44 43 01 00 00 00  ab fe ff ff 01 fd ff ff  |CCDC............|
000002d0  da f5 ff ff 42 54 4e 43  4f 41 54 4c 14 00 00 00  |....BTNCOATL....|
000002e0  a8 d3 01 00 ff ff ff ff  42 54 4e 43 43 53 52 41  |........BTNCCSRA|
000002f0  2c 00 00 00 7c d3 01 00  ff ff ff ff 42 54 4e 43  |,...|.......BTNC|
00000300  43 53 52 47 2c 00 00 00  50 d3 01 00 ff ff ff ff  |CSRG,...P.......|
00000310  74 6f 52 41 01 00 00 00  01 00 00 00 00 01 00 00  |toRA............|
00000320  00 00 ff 00 74 6f 52 47  01 00 00 00 ff 00 00 00  |....toRG........|
00000330  00 ff 00 00 00 00 01 00  6c 43 49 41 03 00 00 00  |........lCIA....|
00000340  7d 13 00 00 c8 00 00 00  00 00 00 00 6c 43 49 47  |}...........lCIG|
00000350  02 00 00 00 7d 13 00 00  c8 00 00 00 00 00 00 00  |....}...........|
00000360  42 54 4e 43 6c 43 54 47  3c 00 00 00 14 d3 01 00  |BTNClCTG<.......|
00000370  ff ff ff ff 42 54 4e 43  53 41 72 50 1c 00 00 00  |....BTNCSArP....|
00000380  f8 d2 01 00 ff ff ff ff  42 54 4e 43 4f 50 50 53  |........BTNCOPPS|
00000390  50 00 00 00 a8 d2 01 00  ff ff ff ff 42 54 4e 43  |P...........BTNC|
000003a0  4e 46 53 4c a4 01 00 00  04 d1 01 00 ff ff ff ff  |NFSL............|
000003b0  42 54 4e 43 53 54 70 53  2c 00 00 00 d8 d0 01 00  |BTNCSTpS,.......|
000003c0  ff ff ff ff 42 54 4e 43  43 54 52 4c 1c 03 00 00  |....BTNCCTRL....|
000003d0  bc cd 01 00 ff ff ff ff  42 54 4e 43 70 53 53 4c  |........BTNCpSSL|
000003e0  a4 01 00 00 18 cc 01 00  ff ff ff ff 42 54 4e 43  |............BTNC|
000003f0  78 52 53 4c a4 01 00 00  74 ca 01 00 ff ff ff ff  |xRSL....t.......|
00000400  42 54 4e 43 4d 50 53 4c  a4 01 00 00 d0 c8 01 00  |BTNCMPSL........|
00000410  ff ff ff ff 42 54 4e 43  43 53 53 4c a4 01 00 00  |....BTNCCSSL....|
[... cut ...]
000004f0  42 54 4e 43 72 50 53 4c  a4 01 00 00 10 30 01 00  |BTNCrPSL.....0..|
00000500  ff                                                |.|
```

I can't really tell you what the values mean, but some data makes sense, for example at offset
0xb8:
```
000000b0  e0 e5 01 00 ff ff ff ff  6d 4e 72 53 46 31 37 58  |........mNrSF17X|
000000c0  38 46 55 58 4b 50 47 44                           |8FUXKPGD        |
```
This seems to be the serial number (see the `SrNm` backwards?). Look at it until it starts to make sense...

jdbkevinma, 2022-02-16:
@nikias 

thanks for your guide, may I know what tool do you use to decode?

nikias, 2022-02-16:
I did this (on macOS):
```
echo "Z2ZDU0AFAAAAAAIAAgACAAAAAABCAAAAI0JMTUYzWDgzNDMwQ1IySlZONTRNRUZSQaUBAAAAffBCpQEAAABj8FVLU1cBAAAAAAAAAEpQAAAAAAAAY2FNVzhTnCqiRwAAAAAAAAAAAABjYU1COFOcNgQIAAAAAAAAAAAAAEJUTkNUQUJDKAEAANj+AQD/////QlROQ0xBQ0IAEQAA2O0BAP////9CVE5DTEFDV/gHAADg5QEA/////21OclNGMTdYOEZVWEtQR0QAAAAAI0dGQy8vLy8qLyoAAAAAAAAAAABCVE5DdHRhQhEAAADM5QEA/////0JUTkNFRkFMZAAAAGjlAQD/////QlROQ3BNUEzsAQAAfOMBAP////9CVE5DbENTUFACAAAs4QEA/////0JUTkNMQ0xDUAAAANzgAQD/////QlROQ2xhQ1RQAAAAjOABAP////9CVE5DbENQRCgAAABk4AEA/////0JUTkNMQ3JQAAEAAGTfAQD/////QlROQ2xDdE2ICgAA3NQBAP////9wU0dDAQAAAAAAAAAAAAAAAAAAACNkb01NVEFYMgAAAAAAAAAAAAAAbmdlUkovQQAAAAAAAAAAAAAAAABHQ0xDAQAAAAAAAAAAAAAAAQAAAFNITEMBAAAAAAAAAAAAAAACAAAAbENvTgAAAAAAAAAAAAAAAAAAAABxU1JQAQAABAAAAAAAAAAAAAAAAEJUTkNOU3RNLAAAALDUAQD/////QlROQ0NsZE1aAAAAvNMBAP////9MQ3hSAAKXOQAAAAAAAAAAAAAALkJUTkNTTUNGEQAAAJzUAQD/////QlROQ0JNQ0ZAAAAAXNQBAP////9CVE5DU01DQhEAAABI1AEA/////0JUTkNCTUNCMAAAABjUAQD/////Q0NWQwAAAACF////NwEAAE0BAABDQ0RDAQAAAKv+//8B/f//2vX//0JUTkNPQVRMFAAAAKjTAQD/////QlROQ0NTUkEsAAAAfNMBAP////9CVE5DQ1NSRywAAABQ0wEA/////3RvUkEBAAAAAQAAAAABAAAAAP8AdG9SRwEAAAD/AAAAAP8AAAAAAQBsQ0lBAwAAAH0TAADIAAAAAAAAAGxDSUcCAAAAfRMAAMgAAAAAAAAAQlROQ2xDVEc8AAAAFNMBAP////9CVE5DU0FyUBwAAAD40gEA/////0JUTkNPUFBTUAAAAKjSAQD/////QlROQ05GU0ykAQAABNEBAP////9CVE5DU1RwUywAAADY0AEA/////0JUTkNDVFJMHAMAALzNAQD/////QlROQ3BTU0ykAQAAGMwBAP////9CVE5DeFJTTKQBAAB0ygEA/////0JUTkNNUFNMpAEAANDIAQD/////QlROQ0NTU0ykAQAALMcBAP////9CVE5DVFRZR2IAAADIxgEA/////0JUTkNsQ3hQkAAAADjGAQD/////QlROQ1JBQ0IyAAAABMYBAP////9CUlRTggAmKvwQpxr0H711AAAAAE1QVFBaMDEzQ2FNYVFhSWhTYVdhYUdpTQQCAQD///z/AQD9/wAAAABhR3BTAgIBAA8AAAAAAAAAAAAAAHRvUkMCAAAAAAD/AAEAAAAA/wAAQlROQ01DU0MoAAAA3MUBAP////9CVE5DTENGTCiUAAC0MQEA/////0JUTkNyUFNMpAEAABAwAQD/" |base64 -D |hexdump -C
```
But I can generally recommend `vbindiff` as a command line hex editor.

truonggiang0710, 2023-08-10:
> is not factory data，is written data in NAND

Why are you aware of this information? Do you have a solution for accessing factory data?

ankit125664, 2024-02-13:
@nikias How to get the SysCfg Data above the iphoneX

jsonlee1688, 2024-04-01:
now  "idevicediagnostics mobilegestalt SysCfg" cmd  returns MobileGestaltDeprecated on ios17,4 any body fixed?

nikias, 2024-04-01:
Apple disabled MobileGestalt query feature. It now only works for `InternalBuild` software, and no replacement is available...

AiXanadu, 2024-04-07:
@nikias Is there any way to get SysCfg data or factory data on the latest iPhone?

CDSQ, 2024-04-20:
> @nikias如何在 iphoneX 上方获取 SysCfg 数据

iailbreak or Ramdisk 
[image]

AiXanadu, 2024-11-16:
> Apple 禁用了 MobileGestalt 查询功能。它现在仅适用于软件，没有替代品可用......`InternalBuild`

Is there any latest progress? I am willing to pay to get the method.

zuodd, 2024-12-05:
[test2.txt](https://github.com/user-attachments/files/18020949/test2.txt)
I have got this data file, but i still dont known how to parse it. I have tried base64 parse, but responsed with the picture:

zuodd, 2025-01-14:
> > Apple 禁用了 MobileGestalt 查询功能。它现在仅适用于软件，没有替代品可用......`InternalBuild`
> 
> Is there any latest progress? I am willing to pay to get the method.

AFAIK, there is no way get factory SN in the latest iPhone, apple has been deprecated any method to access factory data, except old device models or system version < 17.4