libimobiledevice/libimobiledevice #1269: ideviceinfo -q com.apple.mobile.iTunes.store -k AppleID

msk521, 2022-01-25:
any body know why ideviceinfo -q com.apple.mobile.iTunes.store -k AppleID  or  ideviceinfo -q com.apple.itunesstored -k AppleID  is return empty？

mexmer, 2022-01-25:
because apple decided to do so.
ideviceinfo prints out what device return, you can try verbose mode, or if you compiled project with debug info enabled, you can turn on debug mode, and see actual communication with device, but in general.

libimobile library is just interface, so if device doesn't want to give you value, libimobiledevice can't do anything with it.

msk521, 2022-01-26:
> because apple decided to do so. ideviceinfo prints out what device return, you can try verbose mode, or if you compiled project with debug info enabled, you can turn on debug mode, and see actual communication with device.
> 
> libimobile library is just interface, so if device doesn't want to give you value, libimobiledevice can't do anything with it.

Thank you for your reply. I found someone using it " instproxy_browse  iTunesMetadata " or
"ideviceinfo -q com.apple.mobile.iTunes.store -k KnownAccounts --xml" can receive it,but i can't,i want know why

closed 2022-01-26 by msk521

mexmer, 2022-01-26:
> Thank you for your reply. I found someone using it " instproxy_browse iTunesMetadata " or "ideviceinfo -q com.apple.mobile.iTunes.store -k KnownAccounts --xml" can receive it,but i can't,i want know [...]

it's the thing, that sometimes apple decide to change or remove some values. so you can get those things on previous version of iOS, while on newer they don't give you anything. or they just rename value, move them to different domain, or it's just programmer oversight, and it will return in another update.

still there is not much libimobile project can do around it, to know this things, you either need to spend lot of time experimenting, or debugging and dissasembling apple binaries. 

if i knew, i would gladly share what i know, but in this case i really don't know more, than what i can get with experiments.

msk521, 2022-02-07:
> it's the thing, that sometimes apple decide to change or remove some values. so you can get those things on previous version of iOS, while on newer they don't give you anything. or they just rename [...]

Thank you for your reply. I'll try again