libimobiledevice/libimobiledevice #1270: lockdownd_client_new_with_handshake does not return LOCKDOWN_E_SUCCESS

annazolkieve, 2022-01-28:
`lockdownd_client_new_with_handshake` does not return `LOCKDOWN_E_SUCCESS`

```
lockdown_t *lockdown_open(device_t * device)
{
    lockdownd_client_t lockdownd = NULL;
    if (lockdownd_client_new_with_handshake(device->client, &lockdownd,
                                            "pris0nbarake") !=
        LOCKDOWN_E_SUCCESS) {
        ERROR("Unable to pair with lockdownd\n");
    }
```
3GS

annazolkieve, 2022-01-28:
reason:
```
00:47:34 idevice.c:1201 idevice_connection_enable_ssl(): ERROR during SSL handshake: error:0A000152:SSL routines::unsafe legacy renegotiation disabled
00:47:34 lockdown.c:779 lockdownd_client_new_with_handshake(): Session opening failed.
```
and later:
```
00:47:34 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
00:47:34 property_list_service.c:211 internal_plist_receive_timeout(): 369295618 bytes following
00:47:34 property_list_service.c:224 internal_plist_receive_timeout(): received 595 bytes
00:47:34 property_list_service.c:224 internal_plist_receive_timeout(): received 9 bytes
00:47:34 property_list_service.c:224 internal_plist_receive_timeout(): received 7 bytes
00:47:34 idevice.c:689 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -54 (Connection reset by peer)
00:47:34 service.c:167 service_receive_with_timeout(): could not read data
00:47:34 property_list_service.c:229 internal_plist_receive_timeout(): received incomplete packet (611 of 369295618 bytes)
00:47:34 property_list_service.c:231 internal_plist_receive_timeout(): incomplete packet following:
```
how to fix?

annazolkieve, 2022-01-28:
fixed by:
```
		opts |= SSL_OP_ALLOW_UNSAFE_LEGACY_RENEGOTIATION;
		SSL_CTX_set_options(ssl_ctx, opts);
```

mexmer, 2022-01-31:
what version of ssl library you use? it's quite possible they changed "defaults" for handshake, like they did remove support for elder "unsafe" protocol version in some builds.

but yeah, seems this will need to be added to sourcecode, just in case.