libimobiledevice/libimobiledevice #1272: Unable to install libimobiledevice from head on macOS.  brew install libimobiledevice --HEAD

eclaires, 2022-02-01:
Hi,

I see lots of discussion about this error but no simple fix.

11:35:24  ==> Installing libimobiledevice --HEAD
11:35:30  ==> ./autogen.sh
11:36:19  Last 15 lines from /Users/jenkins/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh:
11:36:19  checking for pkg-config... /usr/local/Homebrew/Library/Homebrew/shims/mac/super/pkg-config
11:36:19  checking pkg-config is at least version 0.9.0... yes
11:36:19  checking for libusbmuxd-2.0 >= 2.0.2... yes
11:36:19  checking for libplist-2.0 >= 2.2.0... yes
11:36:19  checking for libimobiledevice-glue-1.0 >= 1.0.0... no
11:36:19  configure: error: Package requirements (libimobiledevice-glue-1.0 >= 1.0.0) were not met:
11:36:19  
11:36:19  No package 'libimobiledevice-glue-1.0' found

We have a script to automate mac setup and it is now failing. I can't do a manual step. How is this solved? Will there be an update so that _**brew install libimobiledevice --HEAD**_ works without complicated work-arounds?

Claire

mexmer, 2022-02-02:
search is here for a reason, see #1217

there are also direct instructions for using script for macos build made by @nikias 

`
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L is.gd/kdT2Nl
bash ./limd-build-macos.sh
`

Diarly, 2022-09-13:
when I curl limd-build-macos.sh
getting error :curl: (7) Failed to connect to is.gd port 80: Operation timed out @mexmer

mexmer, 2022-09-13:
try with https: eg.
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L https://is.gd/kdT2Nl
bash ./limd-build-macos.sh
```
or, since it's shorcut, you can use full url
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L https://gist.githubusercontent.com/nikias/84c79469a1d0f16ff95250f0d51858c3/raw/34de83fe55e7e91735d2d822f50da766fc7a3e1a/limd-build-macos.sh
bash ./limd-build-macos.sh
```
or just download this shell skript from nikias and run it
https://gist.github.com/nikias/84c79469a1d0f16ff95250f0d51858c3/

Diarly, 2022-09-13:
things are working with downloaded limd-build-macos.sh with full url . Thank you. :) @mexmer

Diarly, 2022-09-13:
Hi,

I got  this error  info :

ERROR: build failed for libideviceactivation, check /Users/xxx/limd-build/libideviceactivation_build.log for details.

/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       activation.lo
  CCLD     libideviceactivation-1.0.la
ld: warning: ignoring file /usr/local/opt/libxml2/lib/libxml2.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_htmlNodeDump", referenced from:
      _idevice_activation_parse_html_response in activation.o
  "_xmlBufferCreate", referenced from:
      _idevice_activation_parse_html_response in activation.o
  "_xmlBufferFree", referenced from:
      _idevice_activation_parse_html_response in activation.o
  "_xmlFree", referenced from:
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlFreeDoc", referenced from:
      _idevice_activation_parse_html_response in activation.o
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlGetProp", referenced from:
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlNodeGetContent", referenced from:
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlNodeListGetString", referenced from:
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlReadMemory", referenced from:
      _idevice_activation_parse_html_response in activation.o
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlStrcmp", referenced from:
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlStrlen", referenced from:
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlXPathEvalExpression", referenced from:
      _idevice_activation_parse_html_response in activation.o
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlXPathFreeContext", referenced from:
      _idevice_activation_parse_html_response in activation.o
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlXPathFreeObject", referenced from:
      _idevice_activation_parse_html_response in activation.o
      _idevice_activation_parse_buddyml_response in activation.o
  "_xmlXPathNewContext", referenced from:
      _idevice_activation_parse_html_response in activation.o
      _idevice_activation_parse_buddyml_response in activation.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libideviceactivation-1.0.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

 Could you please help on this?

Diarly

mexmer, 2022-09-13:
are you building on M1 mac?

we will need to ask @nikias for further assistance.

Diarly, 2022-09-14:
@mexmer ,I am building on Intel Core i9 Mac.