libimobiledevice/libimobiledevice #1277: No more working with iOS simulators

xpingc, 2022-02-16:
The ideviceinstaller/idevicesyslog used to work with iOS simulators. But probably starting from XCode 13.2, ideviceinstaller does not work with a simulator.

Is this a known issue or expected?

nikias, 2022-02-17:
Hmm interesting fact; do you have an example how it has been used?

xpingc, 2022-02-17:
I use the following command:

ideviceinstaller -l -u <my_device_udid>

But got the following error:
No device found with udid

I am pretty sure the udid is correct.

The similar thing happens for idevicesyslog.

nikias, 2022-02-17:
Well that is for real devices, but iOS simulators are not (and afaik were never) exposed through usbmuxd so they wouldn't be listed as device (idevice_id -l).

xpingc, 2022-02-17:
All right, so we were saying ideviceinstaller would not work with iOS simulators? Any workaround to make it work with simulators?

nikias, 2022-02-17:
What confuses me is that in your original question you said "used to work with iOS simulators" so I was assuming you knew how to make it work...

xpingc, 2022-02-18:
Probably I remembered it wrong :)