libimobiledevice/libimobiledevice #1282: libimobiledevice-glue-dev not on ubuntu

saday19, 2022-03-06:
relatively cut and dry, was installing dependencies and libimobiledevice-glue-dev is not on ubuntu tried versions 18 and 20, both didnt work. Then looked up the packages on their website, wasnt able to find it. All other packages are there however.

mexmer, 2022-03-07:
glue is not in stable version, unless ubuntu uses github snapshot (which i doubt it does), there will be no glue, or even latest libimobiledevice in them.

saday19, 2022-03-11:
> glue is not in stable version, unless ubuntu uses github snapshot (which i doubt it does), there will be no glue, or even latest libimobiledevice in them.

I dont understand how I am supposed to build it then, when I compile I get an error that says I am missing an included file...

mexmer, 2022-03-14:
download, compile and install libimobiledevice-glue from github, same for libplist, libmobiledevice, libusbmuxd and ubsmuxd