libimobiledevice/libimobiledevice #1284: Question: Why did idevicesetlocation worked for the first time and not after?

markandrewkato, 2022-03-09:
I tried using idevicesetlocation for the first time, it worked without any issues. Next is I tried restarting my device, then retried using idevicesetlocation again but it now asks me to mount a developer disk image.

Just wondering why did it worked the first time and why does it need the developer disk image.

Thanks!

nikias, 2022-03-09:
It needs the developer disk image mounted because the service that can set the location is only available on the developer disk image. You had it mounted the first time; there is no other way this can have worked.

xanadu-lee, 2022-03-22:
Every reboot requires reinstalling the developer plugin.

chloeparker167, 2026-08-25:
The reason idevicesetlocation stopped working after a reboot is that the location‑setting service lives in the developer disk image, which is only present while that image remains mounted; once the device restarts the image is unloaded and the tool can’t find the service. When you need to repeat the operation you must reinstall the appropriate developer disk image, just as you would refresh any missing system component. Checking the [Anderson Court Documents](https://andersoncountycourts.org) for similar version‑dependency issues can be a helpful reminder to verify that all required files are correctly loaded before proceeding.

AiXanadu, 2026-08-25:
How to install developer plug-ins in iOS17 and above?
Currently, the user is relied on to manually activate the developer mode. Is there a way to do this without restarting and without requiring manual operation by the user?

mexmer, 2026-08-25:
AFAIK not possible, it's intentionaly made this way by apple.

in past (ios/ipados 16, watchos 10) you could activate developer mode, on devices without pin. since ios/ipados 17 and watchos 11, it's not possible anymore, it requires user to enable developer mode manually, and confirm it after reboot, regardless devices has passcode set or not.