libimobiledevice/libimobiledevice #1288: Segmentation fault on idevicebackup2 backup with a ro mounted fs over sshfs

C4ne, 2022-03-13:
**Used versions:**
idevicebackup 1.3.1
usbmuxd 1.1.2

**Steps to reproduce:**
- Do a full backup into a directory that will be mounted over sshfs later
- `sshfs -o ro user@yourserver.com:/path/to/backupdir /mnt/backup`
- Then do another backup with: `idevicebackup2 backup --full /mnt/backup`
  - The output will be something along the lines of: 
```
  Backup directory is "/mnt/backup/"
[New LWP 85791]
Started "com.apple.mobilebackup2" service on port 62005.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Enforcing full backup from device.
Backup will be unencrypted.
Requesting backup from device...
Incremental backup mode.
Sending 'f64a137b6d769fd6f7aff7d3c9bb73d7ea042051/Status.plist' (189 Bytes)
Sending 'f64a137b6d769fd6f7aff7d3c9bb73d7ea042051/Manifest.plist' (89.6 KB)
Sending 'f64a137b6d769fd6f7aff7d3c9bb73d7ea042051/Manifest.db' (10.4 MB)
Sending 'f64a137b6d769fd6f7aff7d3c9bb73d7ea042051/Status.plist' (189 Bytes)

Thread 1 "idevicebackup2" received signal SIGSEGV, Segmentation fault.
```

**Tested with:**
- IPhone XR
- iPad Pro (2017)

**Output of gdb:**
```
#0  0x00007ffff7aa7399 in _int_free ()
   from /nix/store/jcb7fny2k03pfbdqk1hcnh12bxgax6vf-glibc-2.33-108/lib/libc.so.6
#1  0x00007ffff7aaaad4 in free ()
   from /nix/store/jcb7fny2k03pfbdqk1hcnh12bxgax6vf-glibc-2.33-108/lib/libc.so.6
#2  0x0000000000408da6 in mkdir_with_parents.constprop ()
#3  0x0000000000408db3 in mkdir_with_parents.constprop ()
#4  0x0000000000408db3 in mkdir_with_parents.constprop ()
#5  0x0000000000408db3 in mkdir_with_parents.constprop ()
#6  0x0000000000408db3 in mkdir_with_parents.constprop ()
#7  0x0000000000408db3 in mkdir_with_parents.constprop ()
#8  0x0000000000408db3 in mkdir_with_parents.constprop ()
#9  0x0000000000408db3 in mkdir_with_parents.constprop ()
#10 0x0000000000408db3 in mkdir_with_parents.constprop ()
#11 0x0000000000408db3 in mkdir_with_parents.constprop ()
#12 0x0000000000408db3 in mkdir_with_parents.constprop ()
#13 0x0000000000408db3 in mkdir_with_parents.constprop ()
#14 0x0000000000408db3 in mkdir_with_parents.constprop ()
#15 0x0000000000408db3 in mkdir_with_parents.constprop ()
#16 0x0000000000408db3 in mkdir_with_parents.constprop ()
#17 0x0000000000408db3 in mkdir_with_parents.constprop ()
#18 0x0000000000408db3 in mkdir_with_parents.constprop ()
#19 0x0000000000408db3 in mkdir_with_parents.constprop ()
#20 0x0000000000408db3 in mkdir_with_parents.constprop ()
#21 0x0000000000408db3 in mkdir_with_parents.constprop ()
#22 0x0000000000408db3 in mkdir_with_parents.constprop ()
#23 0x0000000000408db3 in mkdir_with_parents.constprop ()
#24 0x0000000000408db3 in mkdir_with_parents.constprop ()
...
#260896 0x0000000000408db3 in mkdir_with_parents.constprop ()
#260897 0x0000000000408db3 in mkdir_with_parents.constprop ()
#260898 0x0000000000405cec in main ()
```

xanadu-lee, 2022-03-22:
It looks like the recursive directory creation is not handled properly？

C4ne, 2022-03-22:
Im sorry by I dont know. All I can say is that the backups directory already existed (because I did a --full backup beforehand). Let me know if you need more information.

xanadu-lee, 2022-03-23:
Can you backup via a local directory? i.e. don't use a remote directory.

C4ne, 2022-03-24:
Yeah this works. As soon as I remove the read-only flag, the backup works also over the remote connection. 
But because idevicebackup gave me a segmentation fault I thought it would be important to open an issue on that.

xanadu-lee, 2022-03-25:
You are correct, it should return failure, not crash anyway.

closed 2022-04-08 by C4ne