libimobiledevice/libimobiledevice #1289: /usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_MD_get_size'

18002138971, 2022-03-15:
Making all in tools
make[2]: Entering directory '/home/luhaishan/envirment/ios/libimobiledevice/tools'
  CCLD     idevice_id
/usr/bin/ld: ../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_MD_get_size'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:711: idevice_id] Error 1
make[2]: Leaving directory '/home/luhaishan/envirment/ios/libimobiledevice/tools'
make[1]: *** [Makefile:462: all-recursive] Error 1
make[1]: Leaving directory '/home/luhaishan/envirment/ios/libimobiledevice'
make: *** [Makefile:392: all] Error 2

mexmer, 2022-03-15:
do you have installed openssl development package or properly set path to openssl includes and libraries?

18002138971, 2022-03-18:
yes,i did. my openssl version is OpenSSL 3.0.2 

luhaishan@luhaishan-OptiPlex-7050:~/envirment/ios/libimobiledevice$ openssl version
OpenSSL 3.0.2 15 Mar 2022 (Library: OpenSSL 3.0.2 15 Mar 2022)
luhaishan@luhaishan-OptiPlex-7050:~/envirment/ios/libimobiledevice$

mexmer, 2022-03-21:
i'm spefically asking about development package, not library. make sure, you have only one version of openssl development package installed (either 1.1 or 3.0, but not both).

18002138971, 2022-03-21:
I'm sorry I don't understand very well, you can see the screenshot below, after installation, generate and add the corresponding library to the system path 
[image]

mexmer, 2022-03-21:
no idea what linux distribution you are using, but in most distribution, there are 2 packages for standard libraries, one is with binary (for openssl it will be either openssl, or libopenssl package) and one for development (for openssl it's usually called  libopenssl-dev), if you have dev package installed, usually there is /usr/include/openssl folder, that contains headers for compilation, from content of /usr/local/lib64 i suspect, you have somehow manually installed/compiled openssl 1.1, maybe even 3.0, and it's in conflict with distribution installed openssl 3.0

xanadu-lee, 2022-03-22:
If possible, it is recommended to compile and install openssl 1.1.1
It will work correctly.