libimobiledevice/libimobiledevice #1290: network connect On the win10

18002138971, 2022-03-17:
On my win10 system, the network connection can sometimes recognize the device and sometimes can't recognize the device. Is there any rule for identifying the ios device? If there is a new ios device that needs to be connected to the network, what should I do to ensure stability Identify network devices?

jkcoxson, 2022-05-14:
You can't ensure a device will be detected by usbmuxd. That's just the nature of mDNS.

xanadu-lee, 2022-07-30:
As far as I know, if the device trusts this computer, Apple Device Drive will detect the connection when it is connected to the same local area network. If trusted but not detected, it can usually be found by disabling the network connection (computer) and reopening the network connection.