libimobiledevice/libimobiledevice #1291: "idevicedebug --detach run <bundleID>" would kill and restart the app

josephgiting, 2022-03-22:
I am using the latest version of idevicedebug on iOS 15.4 device:
> idevicedebug --detach run <bundleID>

But this command would kill the specific running app and restart it. 
In my use case, it would be good if _idevicedebug_ could just bring the running app. to the foreground, without killing (and restarting) the app.

Any idea how to do this please?

jkcoxson, 2022-05-14:
I have accomplished this writing my own tool with my [Rust bindings](https://github.com/jkcoxson/rusty-libimobiledevice).

```rust
let debug_server = match device.new_debug_server("jitstreamer") {
            Ok(d) => d,
            Err(_) => {
                warn!("Error creating debug server");
                return Err("Unable to create debug server".to_string());
            }
        };

        let command = format!("vAttach;0000{:X}", pid);

        match debug_server.send_command(command.into()) {
            Ok(res) => info!("Successfully attached: {:?}", res),
            Err(e) => {
                warn!("Error attaching: {:?}", e);
                return Err("Unable to attach".to_string());
            }
        }

        match debug_server.send_command("D".into()) {
            Ok(res) => info!("Detaching: {:?}", res),
            Err(e) => {
                warn!("Error detaching: {:?}", e);
                return Err("Unable to detach".to_string());
            }
        }

        Ok(())
```
This shouldn't be hard to implement into the official tool, but requires having the PID of the process. This is currently blocked by https://github.com/libimobiledevice/libimobiledevice/issues/1161

mexmer, 2022-05-16:
@jkcoxson technically it's not blocked. libimobile provides you with basic connectivity and commands to start service, so you can implement communication with required service on your own no need to wait for library authors to integrate everything.