libimobiledevice/libimobiledevice #1293: ideviceinstaller not working if ipa file path contains Chinese characters on windows

Dantee296, 2022-03-29:
it working totally fine on macos 

ideviceinstaller.exe -i C:\Users\席大大\AppData\apple.ipa -u 46aa83f7a6a3d923e75ba2de26ec9d35163386ae
ERROR: zip_open: C:\Users*席大大*\AppData\apple.ipa: 18*

@qmfrederik @nikias

Dantee296, 2022-03-29:
@josephgiting  can u try it on your side to see if it works i have tried but it doesnt

@nikias its not libzip even stat() method not working

Dantee296, 2022-03-30:
@nikias anyupdate ?
also how can i use wmain in ideviceinstaller 

 int wmain(int argc,  wchar_t **argvc)
 
 undefined reference to `wWinMain@16'

Dantee296, 2022-03-31:
closing ticket , i have resolved the issues it needed modifications in ideviceinstaller to support unicode

```
char * -> wchar_t *
fopen -> _wfopen
zip_open -> zip_open_from_source
main() -> wmain()
```

closed 2022-03-31 by Dantee296

mexmer, 2022-03-31:
yes on windows, when compiling app with unicode support you need to use unicode function, ANSI will have issue with unicode symbols, unless you have set unicode as default coding for system ... which usually you not want.