libimobiledevice/libimobiledevice #1294: How to specify a negative value for idevicesetlocation

mi-mo-sa, 2022-04-01:
idevicesetlocation works for positive values, but fails for negative values.
How should it be specified?
Example: Tokyo (idevicesetlocation 35.685437 139.767438) →OK
Example: NY (idevicesetlocation 40.736306 -73.991533)→NG

nikias, 2022-04-01:
If you check the man page or just run idevicesetlocation, you can see that the usage specifies a `--` after the options to separate it from the options (like UDID). Example:
`idevicesetlocation -u UDID -- 40.736306 -73.991533`

mi-mo-sa, 2022-04-01:
I had misread USAGE.
Your suggestion helped me understand that the "--" was necessary.
Thank you for your response.
====================
Usage: C:\msys64\mingw64\bin\idevicesetlocation.exe [OPTIONS] -- <LAT> <LONG>

closed 2022-04-01 by mi-mo-sa