libimobiledevice/libimobiledevice #1295: No device found, is it connected?

electron1948, 2022-04-03:
Linux Mint 20.2 with iPod Nano 5th Gen
Installed using instructions here:
https://opensource.com/article/21/8/libimobiledevice-iphone-linux
iPod _is_ connected, all I get when entering "ifuse ~/iPhone" is:
"No device found, is it connected?
If it is make sure that your user has permissions to access the raw usb device.  If you're still having issues try unplugging the device and reconnecting it."
Any number of disconnect/reconnects makes any difference.  Any suggestions please?  Many thanks

mexmer, 2022-04-04:
afaik ipod nano is not supported. only ipod touch, because it is running iOS and communicate same way as iPhone.

Abang-L, 2024-04-03:
Same message:
"No device found, is it connected?
If it is make sure that your user has permissions to access the raw usb device.
If you're still having issues try unplugging the device and reconnecting it."

, for iphone6s, ios15.8.
Installed usbmuxd libimobiledevice6 libimobiledevice-utils on linuxmint 21.3.

How to connect this device?

ipa-manager, 2024-12-28:
Same here, trying with a broken iPad, display doesn't work. Trying to find some way of seeing it's screen, preferably with idevicescreenshot. Crostini

mexmer, 2025-01-03:
if your device has usb security enabled, you will not be able to connect it with broken display, because locked devices are not visible on usb, until you unlock device and confirm trust, that's how apple designed it.
i'm reffering to this setting
https://support.apple.com/en-us/111806

AntVil, 2026-10-25:
Hey not sure if this helps anybody, but i managed to fix this by not using my front-usb ports, but rather the ones in the back .. yes i was desperate and it payed off 😄