libimobiledevice/libimobiledevice #1299: Could not perform backup protocol version exchange, error code -1

root-over, 2022-04-15:
I'm trying to backup my iphone with ios 15.4.1, after doing `idevice pair` everything works fine, but as soon as I do `idevicebackup2 backup` it prints me:
```
Started "com.apple.mobilebackup2" service on port 50921.
Could not perform backup protocol version exchange, error code -1
```

Can anyone help me?

nikias, 2022-04-15:
You need to build the latest code yourself, then it will work.

root-over, 2022-04-15:
> 

how can I do?

nikias, 2022-04-15:
If you're on Linux you can try my ubuntu script:
https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh?dl=0

If you're on macOS use this script:
```bash
mkdir -p limd-build;
cd limd-build;
curl -o ./limd-build-macos.sh -L is.gd/3TMOiY ;
bash ./limd-build-macos.sh
```

root-over, 2022-04-17:
> If you're on Linux you can try my ubuntu script: https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh?dl=0
> 
> If you're on macOS use this script:
> 
> ```shell
> mkdir -p limd-build;
> cd limd-build;
> curl -o ./limd-build-macos.sh -L https://is.gd/3TMOiY ;
> bash ./limd-build-macos.sh
> ```

Thank you very much!

jcmonnin, 2022-04-23:
It would be good to have an official release after such an important fix. For users on rolling distros like arch, this will ensure it will find it's way to the official package in a timely manner and avoids the hassle of building and installing on their own.

dan-da, 2023-01-17:
> If you're on Linux you can try my ubuntu script:
> https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh?dl=0

This build script fails for me on ubuntu 16.04 as do the build instructions in the README.

Attaching build log.
[build.txt](https://github.com/libimobiledevice/libimobiledevice/files/10430898/build.txt)

limited-dev, 2023-04-18:
> It would be good to have an official release after such an important fix. For users on rolling distros like arch, this will ensure it will find it's way to the official package in a timely manner and avoids the hassle of building and installing on their own.

Arch user here with the same issue.
I've tried building it from source, which did not work. Using yay did also fail. I'm still quite the beginner using linux, so I don't know what the issue is, but I suspect missing libraries.
I have attached the build output provided by yay. There is probably some easy fix, which I did not find yet.
[bout.txt](https://github.com/libimobiledevice/libimobiledevice/files/11260961/bout.txt)

Thanks a lot in advance.

Blackspell01, 2024-07-25:
> If you're on macOS use this script:
> 
> ```shell
> mkdir -p limd-build;
> cd limd-build;
> curl -o ./limd-build-macos.sh -L https://is.gd/3TMOiY ;
> bash ./limd-build-macos.sh
> ```

For me it says 

ERROR: configure failed for libimobiledevice, check /Users/simon/limd-build/limd-build/limd-build/limd-build/limd-build/limd-build/limd-build/limd-build/libimobiledevice_configure.log for details.

How can I install the latest idevicebackup version on macOS?

mexmer, 2024-07-25:
@Blackspell01 
curl is client app, libcurl is development library. depending on distribution, package containing that might by named either  libcurl, libcurl-dev, or just curl-dev

on debian for example is metapackage libcurl-dev, which tells you either to install libcurl4-openssl-dev, or libcurl4-gnutls-dev (i have libcurl4-openssl-dev installed, because i compiled everything that requires ssl with openssl library)

mexmer, 2024-07-25:
@Blackspell01 are you using latest version of script?
should be this one https://gist.github.com/nikias/84c79469a1d0f16ff95250f0d51858c3
taken from https://gist.github.com/nikias/

unfortunately i don't have arm based mac to test it, on intel works fine.

Blackspell01, 2024-07-25:
Well ty for the hint, in fact I wasn't, but the error still remains. 

```
ld: warning: ignoring file '/usr/local/lib/libtatsu.0.dylib': found architecture 'arm64', required architecture 'x86_64'
Undefined symbols for architecture x86_64:
  "_tss_parameters_add_from_manifest", referenced from:
      _main in ideviceimagemounter-ideviceimagemounter.o
  "_tss_request_add_ap_img4_tags", referenced from:
      _main in ideviceimagemounter-ideviceimagemounter.o
  "_tss_request_add_ap_tags", referenced from:
      _main in ideviceimagemounter-ideviceimagemounter.o
  "_tss_request_add_common_tags", referenced from:
      _main in ideviceimagemounter-ideviceimagemounter.o
  "_tss_request_new", referenced from:
      _main in ideviceimagemounter-ideviceimagemounter.o
  "_tss_request_send", referenced from:
      _main in ideviceimagemounter-ideviceimagemounter.o
  "_tss_set_debug_level", referenced from:
      _parse_opts in ideviceimagemounter-ideviceimagemounter.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [ideviceimagemounter] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```