libimobiledevice/libimobiledevice #1304: idevicesyslog not print log because SSL_read  fail

PuffLady, 2022-04-23:
I found an error when I use idevicesyslog to capture my iphone's log, when I jump to a specific page on my iphone, the idevicesyslog can not print log any more, after download and debug the source file, I found the problem is below code

int r = SSL_read(connection->ssl_data->session, (void*)((char*)(data+received)), (int)len-received); // r is returned -1
int sslerr = SSL_get_error(connection->ssl_data->session, r); // sslerr returned 1

can anyone help me fix it or find a deeper reason

my idevicesyslog version is  1.3.0

PuffLady, 2022-04-23:
above code is in idevice.c, I also tryed master branch version ,still not work

nikias, 2022-04-23:
Some questions.
First of all: Which operating system are you using?
Did you link libimobiledevice against OpenSSL? Or gnutls?
When you built from master, did you make sure it is actually using the library you built instead if the previous 1.3.0 one?
If you're on Linux you can use ldd to find out what is being used:
```
ldd `which idevicesyslog`
```
Will print the libraries used, including libimobiledevice, where you then can run ldd with the path of libimobiledevice library, e.g.:
```
ldd /usr/local/lib/libimobiledevice-1.0.so.6
```
Lastly, what specific page is it you jump to?